OctoPrint/OctoPrint #1653: [Bug] [Fix in fork pending PR] Init script calls LSB function before including LSB functions

tullo-x86, 2016-12-14:
#### Reprodutcion steps:

- On a BeagleBone Black, boot from the [Debian 8.6 IoT image](https://debian.beagleboard.org/images/bone-debian-8.6-iot-armhf-2016-12-09-4gb.img.xz] dated 2016-12-09.
- Login as "root", then  
`adduser fab`  
`adduser fab tty`  
`adduser fab sudo`
- Logout, then login as "fab"
- Clone OctoPrint from git
- Enter OctoPrint's directory and `sudo python setup.py install`
- Edit `~/OctoPrint/scripts/octoprint.default` so that the user is "fab" instead of "pi"
- Set OctoPrint to start on boot:  
`cd ~`  
`sudo cp OctoPrint/scripts/octoprint.init /etc/init.d/octoprint`  
`sudo chmod +x /etc/init.d/octoprint`  
`sudo cp OctoPrint/scripts/octoprint.default /etc/default/octoprint`  
`sudo update-rc.d octoprint defaults`
- Try to start the service with `sudo service octoprint start`
- When nothing happens, check the status with `sudo service octoprint status`
- **Note the reference to Line 31 stating that "log_warning_msg" could not be found**

#### Expected behaviour

The init.d script should successfully start OctoPrint.

#### What happened instead?

The init.d script barfed because it couldn't figure out how to call "log_warning_msg" (because the line which includes it is below the call).

#### Branch & Commit or Version of OctoPrint

1.3.0 tag

______

Yes, I have read the FAQ.

foosel, 2016-12-14:
Thanks for the PR, merged on both `maintenance` and `devel`. Will be part of the 1.3.1 release.

foosel, 2017-01-25:
Part of 1.3.1 which has just been released.

closed 2017-01-25 by foosel