OctoPrint/OctoPrint #1655: Gcode temps disregarded

pearson222, 2016-12-15:
#### What were you doing?

Loading ABS Gcode print files from Makerbot to Octoprint.
 
#### What did you expect to happen?

I expected the print to adhere to the setting programmed in my slicer.  
Tool 0 = 225C 
Tool 1 = 0C
Bed = 115C

#### What happened instead?

Tool 0 = 115C
Tool 1 = 0C
Bed = 0C

#### Branch & Commit or Version of OctoPrint

Version 1.3.0 (master branch)

#### Printer model & used firmware incl. version

Makerbot Replicator 2X
Sailfish 40r33

#### Browser and Version of Browser, Operating System running Browser

Opera 42.0.2393.85 
Mac OS Sierra 10.12.1

#### Link to octoprint.log

http://pastebin.com/fVNRaBKP

#### Link to contents of terminal tab or serial.log

http://pastebin.com/ciHuwAap

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2016-12-15:
@markwal possible gpx or gpx configuration issue?

markwal, 2016-12-15:
Yep. In OctoPrint->Settings->GPX, the gcode flavor is set wrong. It needs to match your gcode (slicer).

Did you mean that you're using Makerbot Desktop to slice? What version?

pearson222, 2016-12-15:
I am using Makerbot Desktop 3.7.0.148.  GPX is set to Makerbot.

pearson222, 2016-12-15:
Everything has been working great for the past 4-5 months.  I have been using Makerbot Desktop to slice and exporting as gcode to Octoprint.  The issue occurred right after updating to Octoprint 1.3

pearson222, 2016-12-15:
The newer versions of Makerbot Desktop (3.7+) are too buggy and prevent me from controlling filament retraction consistently.

markwal, 2016-12-15:
I might have messed up settings upgrade in the GPX plugin somehow.  I have to admit I don't test MakerBot flavor gcode very much (I use RepRap slicers), but I can take a look.  Either way, this is a bug in the GPX plugin (https://github.com/markwal/OctoPrint-GPX), not OctoPrint.  But you should verify the gcode flavor setting in the GPX panel.  If that doesn't work, please enter an issue over there.  You can be terse and reference this issue (OctoPrint#1655)

closed 2016-12-15 by markwal