OctoPrint/OctoPrint #1659: Octoprint 1.3.0 breaks printing to FF Creator Pro

jackpine63, 2016-12-16:
#### What were you doing?

I’m running OctoPi, but they requested me to repost here, that it is an OctoPrint problem.

Printer: Flash Forge Creator Pro dual extruder, Sailfish firmware
GPX: 2.5.5
Slicer: slic3r

After upgrading OctoPi to 1.3.0, when I try to do a print, the temperatures are set on the wrong tool.   For example, printing an already-tested model, gcode sets 65 for Bed and 220 for Tool 1.  However, the first print attempt sets 65 for Tool 0 and 220 for Tool 1.  Nothing for Bed.  Cancel and reprint.  Tool 1 heats to 65.  Then Tool 1 gets  set to 220 and 65 moves to Tool 0.  Again, nothing for bed.

If I back off to 1.2.18, everything is good again.

#### What did you expect to happen?

Bed should heat to bed temperature, extruders should heat to extruder temperature.

#### What happened instead?

Extruder heats to bed temperature.

#### Branch & Commit or Version of OctoPrint

Version 1.3.0

#### Printer model & used firmware incl. version

Late 2015 Flash Forge Creator Pro, Sailfish 7.7

#### Browser and Version of Browser, Operating System running Browser

Safari 10.0.2

#### Link to octoprint.log

https://gist.github.com/jackpine63/08be6cae0b0492a83864271a44c18f15#file-octoprint-20161215-log

#### Link to contents of terminal tab or serial.log

https://gist.github.com/jackpine63/08be6cae0b0492a83864271a44c18f15#file-serial-20161215-log

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2016-12-16:
Sounds like duplicate of #1655 and actually an issue with the gpx plugin by @markwal. OctoPrint itself just pumps through whatever temperatures are in your gcode files, it doesn't touch anything there. But the gpx plugin needs to do translation of these commands so that your printer can understand them and that is where things might go wrong. Please report this on the plugin's bug tracker: https://github.com/markwal/OctoPrint-GPX/issues

foosel, 2016-12-16:
Or even better, take a look at this ticket: https://github.com/markwal/OctoPrint-GPX/issues/32

jackpine63, 2016-12-16:
Sounds like same symptoms.  IDK, but Octoprint works perfectly with the same physical gcode file at version 1.2.18, but exhibits these symptoms at 1.3.0.  Maybe one of the API changes resulted in the problem?

foosel, 2016-12-16:
I can't comment on that because I don't know well enough how the gpx plugin works internally, especially with regards to configuration management what appears to be the problem here. In general all documented APIs should have stayed stable, but some plugins need to go beyond those to do their thing and then all bets are off. I'm pretty sure that @markwal has been keeping track of what was changed in 1.3.0 over the course of its development, so if it was a general incompatibility problem he'd have noticed and told me sometime between 2015 and now. Must be some corner case.

markwal, 2016-12-16:
Yes, I generally run on the devel branch so I can see stuff coming.

Since a save settings and reboot solved it in at least one case, I wonder if there is a transient state after config.yaml upgrade where the plugin isn't seeing It's settings. It may be that I'm accessing the settings incorrectly.

gpx gets the setting in question from gpx.ini in the plugin data folder.  So maybe I'm discovering the folder incorrectly.  I'll take a look.

foosel, 2016-12-16:
@markwal On first look, that actually looks correct what you are doing there. I also can't remember changing anything about the data folder between 1.2.x and 1.3.0. What has changed however is the rather aggressive caching on stuff like the settings API in order to reduce the load on the backend. Could something like that maybe be the reason here, causing some value to be set incorrectly on settings save or something like that? Just wild guessing.

foosel, 2016-12-16:
@markwal one thing that did change is that the frontend will no longer sent the full data but only what changed. That was actually something that via the API was always possible throughout OctoPrint, but the frontend didn't utilize it. If no settings for a plugin are contained in that changed data, the plugin's settings saved handler also won't be called. From your code, that would mean that `self.printer.refresh_from_ini` would not be called if none of the actual plugin settings as returned by your plugin's `on_settings_load` had changed on the dialog save. Since you are writing the ini on settings save from the frontend code via a blueprint endpoint, I'm actually surprised that this blueprint endpoint takes care of the ini reload. Could this maybe be the issue, that the ini gets written on settings save, but the printer not properly reconfigured? Then again, I would expect things to only cause issues then on very first configuration, which shouldn't be the case for existing installs.

jackpine63, 2016-12-17:
Based on what you have above, problem solved.  I didn't even think to check GPX settings.  Apparently, the upgrade from 1.2.18 to 1.3.0 reset the GPX settings to default (?)  I went in to my 1.2.18 backup and found: 

$ ls -l
total 16
-rw-r--r--  1 pi  pi  171 Nov 11 17:37 gpx.ini
-rw-r--r--  1 pi  pi  171 May 27  2016 gpx.ini.bkp

Both files are identical.  However, in the upgraded image, I found the two files were different.    gpx.ini.bkp matches the two files in the 1.2.18 backup:

[machine]
actual_filament_diameter=1.75
[printer]
build_progress=0
ditto_printing=0
gcode_flavor=makerbot
machine_type=fcp
nominal_filament_diameter=1.75
recalculate_5d=0

However, the active gpx.ini in the 1.3.0 system is different:

[machine]
actual_filament_diameter=1.75
[printer]
build_progress=1
ditto_printing=0
gcode_flavor=reprap
machine_type=r1d
nominal_filament_diameter=1.75
recalculate_5d=0

Note the different values for gcode_flavor and machine_type.  I guess I could have updated in Octoprint, but I just copied gpx.ini.bkp to gpx.ini and rebooted.  Things are back to working correctly now.

I didn't go back and try to repeat the upgrade to verify I can duplicate the problem (out of spare SD cards), but appears after the upgrade one has to go into GPX settings and reset everything, or restore the file on the file system as I did.

Thanks

markwal, 2016-12-17:
Thank you @foosel for looking into it so far, I didn't expect you to have to spelunk in the plugin.

Also, thank you @jackpine63 for checking that. It'll save me a bunch of going back and forth to get a repro, I think.  I really appreciate you coming back to update with that info.

Ok, I think I've enough to find this thing and fix it, now I just need to finish this gig I'm working on so I can have a minute to look at this.

jackpine63, 2016-12-18:
Thanks.  If you need me to try anything, let me know.  I haven't made any changes in 1.3.0 other than fixing gpx.ini, so I could restore 1.2.18 over top of 1.3.0 and re-upgrade if that would help.

foosel, 2017-01-27:
@markwal what's the status of this ticket? Do you track it in your plugin's bug tracker? I'd actually like to close it since I really can't do anything about it from my side as far as I see right now.

markwal, 2017-01-29:
Makes sense.

closed 2017-01-29 by markwal