OctoPrint/OctoPrint #1661: [Request] Trigger PrintDone only after running afterPrintDone script

eyal0, 2016-12-16:
#### What were you doing?

Added an afterPrintDone Gcode script to run after each print to position the platform in front of the webcam.  Also added a PrintDone event to take a photo and email it to me.

#### What did you expect to happen?

I expected the afterPrintDone Gcode to run after the file was finished printing but before the Status changed to "Operational".  Then I expected a PrintDone event and then the camera to take a photo and send an email.

#### What happened instead?

The Gcode and the camera started at the same time.  The photo was taken just as the platform was starting to move instead of waiting for the afterPrintDone script to complete.  The email received had a photo from before the platform was in the right place.  The platform ended up in the right place, however.

#### Branch & Commit or Version of OctoPrint
1.3.0 (master branch)

#### Printer model & used firmware incl. version
Lulzbot Mini

#### Browser and Version of Browser, Operating System running Browser
Google Chrome Latest

I have read the FAQ.

Maybe the before and after Gcode could be sent as part of the file to print?  A good solutt ion should work for SD card and for local files.  Maybe when the printing is complete, instead of triggering the PrintDone event, immediately start the afterPrintDone script and only when it's finished, then trigger PrintDone.  The status should change from Printing to Operational only after the script has completed.

derpicknicker1, 2017-01-04:
The name of the script says everything: -AFTER-printDone. Not beforePrintDone and not afterPrint. This is very clear and there is no need to change. It's the right order as the names tell you.
The only thing to discuss is: are more trigger points for script snippets needed?

foosel, 2017-01-09:
What @derpicknicker1 said. Also, it's not possible to change things like that without possibly breaking behaviour of existing plugins and scripts out there - basically maintaining backwards compatibility is crucial.

Events are also processed asynchronously - that is necessary in order to prevent long running event handlers from stalling the whole system. Additionally your firmware processes move commands asynchronously. Which makes synchronizing a "move end" on your printer with a processed event tricky if not impossible.

What might work is injecting an after print done script that does your movement, adding an `M400` to that to wait for said movement to finish, then after that adding some custom trigger command (`@triggerMyPlugin`) and adding a gcode handler such that said command gets never sent to the printer but actually triggers your picture taking and emailing instead. That way OctoPrint will wait for the movement to finish (`M400` blocks and only returns `ok` once all queued moves are done), then trigger your plugin through that custom command. No event handler needed.

eyal0, 2017-01-09:
Sure, a way to send commands to octoprint would work great, too!  In fact, I wouldn't even need a "triggerMyPluggin", I would just like to be able to add this into my gcode:

(@systemCommand "/bin/sh -c curl...")

And octoprint should intercept it and run the command without passing it to the printer.  Other CNC controllers have things like this already.  This would solve my problem.  It would also provide a general solution that combines scripts to run before/after and gcode to run before/after.

Something like this doesn't exist yet?  Does the octoprint framework have a way to at least make a plug in to intercept gcode and do stuff like this?

foosel, 2017-01-09:
Yes, plugins can do that, look into [the octoprint.comm.protocol.gcode.* family of hooks](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase).

I'd rather not add the capability to execute arbitrary system commands through GCODE though (not even through a plugin), that just screams for being exploited in some way.

foosel, 2017-01-27:
Closing since AFAIK there are no questions left open (if not, please comment!) and as outlined above the current behaviour cannot be changed without breaking existing third party code - and even with the change the expected outcome would not be achieved thanks to the asynchronous nature of events and firmware movement planning.

closed 2017-01-27 by foosel