OctoPrint/OctoPrint #1663: [Request] Print Speed Adjust on the Fly

dcwalmsley, 2016-12-17:
After uploading and slicing STL file, I find myself needing to adjust the print speed while printing.  Looking for a slider function added under Control tab that allows adjust speed in percentage with 100 percent being the default speed allowing plus or minus 50% (ie, 50-150%).

BerndJM, 2016-12-18:
There is already a slider "Feedrate" on the control tab - isn't it that what you're looking for?

dcwalmsley, 2016-12-18:
No.  Looking for a means to increase or decrease mm/s printer speed.  
Instead of terminating a partial print and reslicing at a different speed 
I'm looking to make print speed adjustments on the fly.

BerndJM, 2016-12-18:
> Print speed adjustments on the fly
Call me stupid, but thats exactly what this slider does ...

dcwalmsley, 2016-12-18:
Oh...  OK.  I thought it had a different function.  Maybe reword it to 
Print Speed vice Feed Speed.

I apologize for my ignorance.

foosel, 2016-12-18:
As already pointed out, that's already there. Renaming it would make it inaccurate - it is a feed rate modifier.

closed 2016-12-18 by foosel