OctoPrint/OctoPrint #1666: Timing out in middle of print

simplyfloid, 2016-12-20:
I have successfully printed my object twice inbetween many failures. Timing out for no apparent reason is my current issue that I have seen several times in the 1-2hr mark of a 4hr print. Additionally, octoprint has always been semi-buggy for connecting in the first place, sometimes it takes several octoprint reboots to recognize printer. Sometimes just disconnecting from the auto-connect that seems to stall, and then connecting manually will work. Once I connect though, it seems to work until this recent string of stopping prints in the middle and timing out.

Monoprice Maker Select V2.1
Octopi image on pi zero with HubPiWi module providing usb ports and wifi
C615 Webcam powered by pi (500ma draw)
Pi powered by 3" cord into 2.0a block

Last Lines of Terminal:

Send: N43240 G0 F9000 X172.732 Y149.425*75
Recv: ok 43240
Send: N43241 G1 F1200 X174.259 Y147.898 E2523.33999*63
Recv: ok 43241
Send: N43242 G0 F9000 X174.259 Y148.464*67
Recv: ok 43242
Send: N43243 G1 F1200 X172.732 Y149.990 E2523.39384*56
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1767
Changing monitoring state from 'Printing' to 'Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1767'
Connection closed, closing down monitor

octoprint.log file:

[octoprint.txt](https://github.com/foosel/OctoPrint/files/664813/octoprint.txt)

LeftyBC, 2017-01-08:
I had this issue with OctoPi, changing to a better USB cable for my Monoprice Select Mini solved it.

foosel, 2017-01-09:
```
Changing monitoring state from 'Printing' to 'Offline: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:1767'
```

Your printer went away. There's nothing OctoPrint can do about that. Usually these kinds of issues are caused by bad USB cables (as suggested by @LeftyBC) or an insufficient PSU on the Pi (if a Pi is used). 

See also the length discussion in #1603, which this ticket actually is a duplicate of.

closed 2017-01-09 by foosel

bevangg, 2017-08-25:
I had this exact issue. I have eliminated power supply problems, usb cable problems, Pi Zero W problems and a few other things that I felt could have been a cause. As Foosel has pointed out so many times when this issue raises its head it is not octoprint. It is a coms issue. 
In my case it turned out to be a faulty Mega. I had a spare lying around so swapped it and all working fine now. If I ever find the time I might try to work out what was wrong but reckon it is the Atmega8 serial chip.