OctoPrint/OctoPrint #1668: [Request] Configurable scrollback buffer size for terminal

twilde, 2016-12-20:
There are times when it would be really helpful (when doing extensive debugging work on a printer, for example) to have a larger scrollback buffer in the terminal than 300 lines. I would imagine the default could even safely go up, but at some point it will have performance implications; if this could be configured either per-instance or even during a session it would be very helpful.

ntoff, 2016-12-24:
Isn't that what enabling the serial log is for? If you go into settings in the "serial connection" category there's an option to enable the serial log. Pretty sure it just logs everything that happens in the terminal out to a text log file that you can scroll through at will and will continue logging until you turn it off, semi infinite scrollback if you will.

twilde, 2016-12-26:
That setting is behind a big "may hurt performance" warning, and also available for admins only; I'd like my users to be able to select this when needed rather than having it on all the time (and then I'd have to figure out a way for them to access serial.log too).

ntoff, 2016-12-28:
I agree and would kind of like a "serial log per print" option like the timelapse setting where if I have a print giving me trouble (like recently a print stopped half way) I could tick a box, or press a button to "print and log" for *just* that print. Turns out in my case it was just a thermal runaway because the cooling fan cooled the bed too far so no issue ticket for you foosel, but i had to go sifting through the serial log to find it which I just leave on and don't seem to experience much if any detrimental performance hits. Then again I print fairly slowly so I don't have data streaming off the sd card all that quickly. Probably something easily added via a plugin.

petonic, 2018-09-09:
Here's a use-case where a larger terminal buffer would be helpful. Turns out that the MonoPrice Mini Select's firmware apparently doesn't calculate averages for the PID that well.  I've written a Python script to calculate it correctly, and it relies on either the terminal text, or the serial log.  I don't want to configure the latter, for performance reasons.

For a run of 15 PID cycles, sometimes 300 lines isn't enough.  Even if I suppress spurious output, those suppressed lines seem to count against the 300 line buffer limit.

Would also be nice, but realize that it's dangerously close to creeping featurism, to be click a button and copy all of the output from the terminal buffer to the clipboard... :-)

foosel, 2018-09-14:
> [use case]

To be honest, that sounds like a custom plugin handling that specific case would be way more useful than thousands of lines of scrollback (which btw can cause *severe* performance issues in the browser).

> Would also be nice, but realize that it's dangerously close to creeping featurism, to be click a button and copy all of the output from the terminal buffer to the clipboard... :-)

[image]

;)

petonic, 2018-09-14:
😳 on the "Copy All" button.

Thanks, Gina.  Well, we don't want to introduce delays for a special one-off case.  I may look into developing a plugin, although I haven't done so yet.

JJJpnt, 2018-09-21:
I totally agree that would be a nice option. For instance, when I do delta calibration, I don't care about browser performance issues at all, though I would like to be able to do drink a coffee, coming back and simply see the feedback in the terminal. So I'd like to increase it say to 3000 only to take it back to 300 afterwards (or only keeping unfiltered lines would be a nice option too - but it'd be getting too complex I guess). Put a huge red waring label on it, but I would really love having that as a setting. 300 is really, really short sometimes.

PCChazter, 2023-09-21:
> the 'Suppress' checkbox under the terminal needs to become an 'Ignore' checkbox

I found a way to get this done. Probably could be put into a plugin, but at this point, this works for me:

Using a terminal on the Pi or computer running Octoprint, run

`nano ~/OctoPrint/venv/lib/python3.10/site-packages/octoprint/static/js/app/viewmodels/terminal.js`
 
and right before where it says (line 271 in Octoprint 1.9.2)
```
            if (self.autoscrollEnabled()) {

                // we only keep the last <buffer> entries
                newLog = newLog.slice(-self.buffer());
            }
            self.log(newLog);
            self.updateOutput();
```
add:
```
           /*Added at line 271 in file ~/OctoPrint/venv/lib/python3.10/site-packages/octoprint/static/js/app/viewmodels/terminal.js
            to make terminal regex the whole log instead of just what is visible. This will help the buffer from clearing what you
            want to see because of suppressed messages.
            Snippet copied and modified from lines 59-79*/
            if (self.enableFancyFunctionality()) {
                var regex = self.filterRegex();
                var lineVisible = function (entry) {
                    return regex === undefined || !entry.line.match(regex);
                };

                var result = [];
                var lines = newLog 
                _.each(lines, function (entry) {
                    if (lineVisible(entry)) {
                        result.push(entry);
                    } 
                });

                newLog = result;
            }
            /*end of add*/
 ```           

restart Octoprint, and it will be changed to not use up the buffer with suppressed messages in the terminal.

arunderwood, 2024-01-10:
I ran into this issue while doing a [PID Autotune](https://marlinfw.org/docs/gcode/M303.html).  The tuning completed and printed out the new PID values but the suppressed status and temp messages quickly pushed those lines out of the buffer even though they were suppressed.

I ended up having to run the tuning a few times and sit by the printer so I could capture the values before they were pushed out.

Raspberry Pis have gotten a lot more powerful since this Issue was opened.  Maybe the terminal output could be buffered server-side and streamed to the webUI?

kubedzero, 2025-12-09:
I wanted to +1 and bump this.

https://github.com/OctoPrint/OctoPrint/blob/5229c24e4c8a0addb69af67d46186631685e9b44/src/octoprint/static/js/app/viewmodels/terminal.js#L18

I found the place where `ko.observable(300)` and `ko.observable(1499)` are set, which correlate with the `self.buffer` and `self.upperLimit`, which translates to the `gettext("showing %(displayed)d lines (buffer full)")`. I wanted to put in a request to allow customization of these values, as I sometimes filter so many things out that the 300 line buffer fills up and I can barely see the lines I don't want filtered out, and sometimes I scroll back and want more than 1500 lines to current printouts. 

I wanted to submit a feature request for octoprint to add this to some sort of configuration file, but I see in this issue that it's already somewhat been requested. To solve, I'd love to see the value either manually editable in a config file in the codebase, or (better) via the webgui. Maybe it could be added to the Printer->Terminal Filters settings page?

Video of the issue with the 300 line limit causing unfiltered messages to pop up for only a couple seconds. This is a Benchy printing at 200mm/s, so faster speeds would have even bigger issues with this.

https://github.com/user-attachments/assets/7123b452-c945-4227-a0bd-f24e2e507fa7

I'm hoping that computers these days are powerful enough to take this on.

In the meantime, I guess I'll edit these values manually in the `terminal.js` file on my instance