OctoPrint/OctoPrint #1669: [Request] Avoid timeout when retracting too much filament (swaping)

docteurzoidberg, 2016-12-21:
#### What were you doing?

I try to change my filament spool before printing.

To do so, I do multiple steps:

- First i use the temperature panel to heat up the nozzle at 190 for my PLA. then I wait for temperature to reach correct value.

- Then I use octoprint retract control in "Control" tab, to retract a large amount (90-100 centimeters)

Since the extrude command is very long for that length (I believe I should add a custom button which set the extrude command at higher speed), Octoprint seems to loose connection with serial port.

Few questions here: 

1) Should I propose a custom "Filament changing" plugin implementation, which will handle timeout, and custom retract speed / length ? or should it be integrated in the "Control" tab ?

2) Should the timeout can be handled in Octoprint when command is done manually (via the control tab) ?

Thanks for the great job. I do code (with some skills in knockout.js), so I will try to contribute as much as I can on this wonderful project, pull requests incoming :)

---
I use octoprint 1.3.0 latest stable image, with a FLSun3D printer (kossel delta mini like), MKSBoard 1.5 electronics (arduino mega 2560).
---

ntoff, 2016-12-23:
Are you using just a regular G1 E-100 or an actual filament change command handled by your firmware?

Maybe you could try firmware retract http://reprap.org/wiki/G-code#G10:_Retract and add that to octoprint's list of long running commands?

docteurzoidberg, 2016-12-23:
Yes I actually use the "retract" button of Octoprint which send a regular G1 command. 

My question is about the needs of implementing it (or not?) with a brand new button, directly in Octoprint control tab. (which could send the G10 and ignore timeouts about it if the firmware handle it)

I know it can be implemented with a custom button, the question is more about why not implementing that core feature in Octoprint directly.

ntoff, 2016-12-24:
Also you said centimetres 

>large amount (90-100 centimeters)

do you mean millimetres or are you actually retracting an entire metre of filament? That is a lot of filament to be unloading via retract. I can extrude and retract 100 *millimetres* without triggering a timeout.

docteurzoidberg, 2016-12-24:
yes I actually retract about a meter (100centimenters) of filament (bowden extruder on a delta printer)

And yes, the question was about control to do filament changing, so both "Retract" & "Loads" controls. You saw my point ;)

(Sorry for my bad english if i'm not that clear :D)

VBen, 2017-01-02:
@docteurzoidberg 
what firmware are you using?
I'm using marlin and use the command [M600](http://reprap.org/wiki/G-code#M600:_Filament_change_pause) to change the filament.

docteurzoidberg, 2017-01-04:
@VBen: you use it both for unloading and loading ? I don't see any speed parameter in the M600 also (retracting 1 meter is long at "normal" speed)

Btw, I actually "solved" my problem with two gcode scripts, one for loading and one for unloading, which each do multiple extrude/retract commands with correct speed and low timeouts.

*These scripts could be easily generated by a plugin, which will take filament distance to from extruder to hotend, and speeds into account)

docteurzoidberg, 2017-01-04:
My two scripts for example:   (load/unload 95cm length, hotend at 195°, without timeouts)
filament loading: http://pastebin.com/nnQdWbEa
filament unloading: http://pastebin.com/wzgmdhc3

VBen, 2017-01-04:
@docteurzoidberg 
It so currenly ony configurable vis the marlin sourcecode
[See configration_adv.h](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration_adv.h#L680)

foosel, 2017-01-09:
I find it weird that OctoPrint is running into a timeout here to begin with. `G1` is a buffered command. Which means that when you send it to the firmware, it will get acknowledged by the firmware when it gets pushed into the planner buffer - long before it finishes executing.

The distance value should not matter at all, unless all slots in the planner buffer are completely filled up the `G1` retract should pretty much instantly trigger an `ok` from the firmware, after which regular `M105` ping pong will take up again.

I feel like something is still missing here since it apparently is not actually doing that. Could you provide the terminal output for the original issue?

foosel, 2017-01-27:
@docteurzoidberg still waiting for the terminal output for the original issue :)

foosel, 2017-02-24:
No terminal output, no way to solve this properly :(

closed 2017-02-24 by foosel