OctoPrint/OctoPrint #1671: Unable to create folders in OctoPrint Web Console

lastrogue, 2016-12-23:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box). Do not delete any
lines from the template but those enclosed in [ and ] - and those please
DO delete, they are only provided for your information and removing them
makes your ticket more readable :)

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

Salandora, 2016-12-23:
Hello,

try this:
Click the Wrench and unselect the Hide empty Folders option.
This should fix your problems, please report back.

ntoff, 2016-12-23:
>Click the Wrench and unselect the Hide empty Folders option.

I was about to comment and say creating the folders works fine here using those steps (using the maintenance branch) then saw this. Why is that even an option? I mean I get it, you don't want empty folders cluttering up the UI but as soon as you create a folder it's going to be empty and thus invisible. Hiding successful prints I get, upload a bunch of stuff to print and as it gets finished it automatically gets hidden but why hide folders? Maybe there needs to be a notice of like "15 files (9 hidden), 3 folders (2 hidden)" somewhere in that file list just to let people know that stuff IS there, it's just hidden.

Salandora, 2016-12-23:
@ntoff I think you're right about the notice this would be helpful, and yes the hide empty folders filter is basically to not clutter the small files list section with empty folders.
And maybe the hide empty folders option should not be enabled by default, because this leads to exactly this. People think it'll not working.
Hmm @foosel what do you think about this?

lastrogue, 2016-12-23:
@Salandora That did the trick, and sure enough I was able to see all the different folders I had created! Thank you for pointing that out. And I agree with @ntoff that a notification would be helpful pointing out x files hidden and y folders hidden, or something to that effect.

foosel, 2017-01-09:
> And maybe the hide empty folders option should not be enabled by default, because this leads to exactly this.

Uhm. It isn't, for this reason actually. From a fresh upgrade to 1.3.0:

[image]

But I like the idea of a "n files (m hidden), x folders (y hidden)" display. I'm a bit at a loss though were to stuff that...

mrardon, 2017-01-09:
I was bitten by this as well, I had changed it to hide the folders but forgot about the setting. 

Not knowing all the space restrictions/usage but above the Free/Total maybe? Could be as simple as "N hidden" if you don't want to be verbose.

jaustindavid, 2017-02-20:
+1 bitten by this.  I have two octoprint servers, and one of them (older, 1.3.0 at creation) got this option.  I have no idea how.  I definitely was able to create a fair series of folders, then "lost" the ability (due to this feature).

foosel, 2017-02-21:
Too many people bitten by this combined with this sudden "uh, what value does this actually add feeling" equals removed filter.

Also I think my above comment claiming it was disabled by default was wrong (and probably falsified by some left over bits in my browser profile). Anyhow - it's gone now on `maintenance` and soon `devel` and that will be part of the 1.3.2 release.

foosel, 2017-03-16:
1.3.2 was just released. Funnily enough I managed to remove that option, but left the context menu entry for it 🤦‍♀️ . So while that shouldn't bite people any more, we now have a context menu entry that does nothing. Stupid, will be fixed in 1.3.3.

foosel, 2017-03-16:
And already solved on `maintenance`, soon `devel`, to be released with 1.3.3. Again - stupid ;)

foosel, 2017-05-31:
1.3.3 was just released.

closed 2017-05-31 by foosel

OutsourcedGuru, 2017-12-18:
What's interesting about this from a UI standpoint is that if—upon toggling the ShowEmptyFolders feature ON when creating a new folder—the user may suddenly see many (surprising) new (previously-empty) folders as a result of this programming.

With respect to usability, upon creating a new folder it may be worth instea just sending a notice/alert to the screen to remind them that the option is currently toggled OFF and where they may find it.

foosel, 2017-12-18:
The option was removed in 1.3.3 ;)

OutsourcedGuru, 2017-12-18:
Honestly, [I personally adjusted](https://github.com/OutsourcedGuru/roboc2-pi-samba-upgrade) my /etc/samba/smb.conf file and then I'm able to push files to my Octoprint via Finder (OSX) via...

**Cmd-K** then **smb://pi@charming-pascal.local/sharename**

This seems to work quite well. It requires you to press the Refresh button within the OctoPrint -> Files area to see the update.