OctoPrint/OctoPrint #1672: Picture from Raspicam

manjoel, 2016-12-23:
Hallo,
I'm trying to connenct an raspi cam to octoprint

I followed this tutorial to install octoprint https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian
When I'an using this command ./mjpg_streamer -i "./input_raspicam.so -fps 5" -o "./output_http.so" , the light on the cam is activated but no picture the MJPG-Streamer Demo Pages is working
Starting octoprint the light of the raspicam is not activ.

Using this command ./mjpg_streamer -i "./input_file.so -f plugins/input_testpicture/pictures/ -e" -o "./output_http.so -w ./www -p 8080" I have the demo picture in octoprint.

Ever with the prebuild octoprint I have no picture from the raspicam.

And what should I do the have an automatic activation of the cam

greeds

camnewnham, 2016-12-30:
Have you tried "raspistill -o image.jpg" This will tell you the state of your pi camera id it doesnt work. If its not enabled, you can turn it on in "sudo raspi-config" / interfacing / camera.

But I believe this question belongs on the G+ forum or other, as it doesn't seem you read instructions when you went to post this issue...

manjoel, 2016-12-30:
It was the firmware of the raspberry, after an upgrade every thing worked

VBen, 2017-01-02:
@manjoel 
you should close this issue

closed 2017-01-07 by GitIssueBot