OctoPrint/OctoPrint #1674: [Request]Disallow downloads of gcode files to non admins and non users

ntoff, 2016-12-25:
Some slicers allow post processing scripts which can upload a file directly to octoprint through curl / windows powershell / etc., and will leave the post processing upload command embedded inside the gcode file, including the API key!!!!!!!. *cough* Simplify3d *cough*

example extract from an uploaded gcode (api key and my PC username redacted)
https://gist.github.com/ntoff/c37e43bfb2bd9352c0c5a182dd114f73

I had a go at disabling the download button for people who aren't logged in but is most likely not the right way of doing it (though it does seem to work but I haven't tested it with access control disabled so I won't post my "fix" here (yet)).

While this is probably a low risk security issue and may only be limited to users of simplify 3d who use the post processing script to upload directly to octoprint, it would be nice to have non logged in persons be unable to easily access files just because, and it brings things more into line with the delete, load, and load and print buttons which are disabled for non users.

Should this be a feature flag, disallowing access by default? Is this part of any planned future granular permission control? Perhaps allowing access to logged in users and admins, but not anonymous access?

ctgreybeard, 2016-12-26:
I'm assuming that you are using curl in your post-processing scripts ... use the curl -K option and create a curl config with the API key in it. The only reason S3D includes the API key is because you put it there.

ntoff, 2016-12-26:
no, I'm using powershell

ctgreybeard, 2016-12-26:
I don't know powershell but my comment still applies. You have entered the key on the command line I suspect and that is why Simplify3D includes it in the gcode. If you can point me to where you found the technique I can probably suggest a more secure alternative.

closed 2016-12-26 by ntoff

foosel, 2017-01-09:
Disabling the download of GCODE files will be part of the implementation of #1110 

That S3D happily embeds the used shell code is troubling news. I'd raise that issue with them. Then again, they've been pretty much completely unresponsive with regards to OctoPrint integration (I tried), so I'd be pretty surprised if they'd react to this.

Nj-BigMike, 2018-01-16:
The api being embedded is not the only reason that downloading files should not be allowed. Truthfully the entire file section should not show up for a non-admin or at least non user.