OctoPrint/OctoPrint #1675: Erroneous Warning Message: Object doesn't fit print volume

SparseTheory, 2016-12-26:
#### What were you doing?
Loading gcode print (gcode was generated by Simplify3D)

#### What did you expect to happen?
Print to load without warning message (bounding box of object is 70mm x 70mm x 70mm - while my printer has capacity of 200x200x200. (see attached file)

#### What happened instead?
Warning message: 
Object doesn't fit print volume
Object in stool.gcode exceeds the print volume of the currently selected printer profile, be careful when printing this.

More 
Object exceeds print volume in depth.
Object's bounding box: (0.00, 0.00, 0.00) × (195.45, 245.75, 120.81)
Print volume: (0.00, 0.00, 0.00) × (200.00, 200.00, 200.00)
You can disable this check via Settings > Features > "Enable model size detection [...]"

#### Branch & Commit or Version of OctoPrint
1.3.0

#### Printer model & used firmware incl. version
xyz DaVinci - which was flashed with Repetier firmware

#### Browser and Version of Browser, Operating System running Browser

Chrome - Version 55.0.2883.87

#### Link to octoprint.log
(error does not show here)
2016-12-26 03:34:59,534 - tornado.access - WARNING - 404 GET http://www.baidu.com/favicon.ico (127.0.0.1) 16.13ms
2016-12-26 04:21:18,202 - tornado.access - WARNING - 401 POST /api/login (127.0.0.1) 26.29ms
2016-12-26 04:21:41,191 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.178
2016-12-26 04:24:19,528 - tornado.access - WARNING - 404 GET /cgi/common.cgi (127.0.0.1) 12.86ms
2016-12-26 04:24:19,984 - tornado.access - WARNING - 404 GET /stssys.htm (127.0.0.1) 12.49ms
2016-12-26 04:24:22,448 - tornado.access - WARNING - 404 POST /command.php (127.0.0.1) 22.54ms

#### Link to contents of terminal tab or serial.log
n/a

#### Link to contents of Javascript console in the browser
See screenshot
[image]

#### Screenshot(s) showing the problem:
See screenshot

[stool.zip](https://github.com/foosel/OctoPrint/files/672529/stool.zip)

SparseTheory, 2016-12-26:
Please see updated ticket.

JohnOCFII, 2016-12-26:
@SparseTheory - Do you have any priming activity happening outside the default print area?  The Prusa i3 MK2 primes outside the standard print area, and thus needs a separate custom bounding box defined in the printer profile area.

SparseTheory, 2016-12-26:
No - While the printhead is parked slightly outside the print area, the
bounding box covers that area.  The only negative moves (outside the box)
are relative moves at the very end of the print and would be well within
the limits.

foosel, 2017-01-09:
Something in your file makes OctoPrint's GCODE interpreter think that an extrusion happens at Y=245.75mm, which is outside your print volume. Which in turn triggers the message.

On first look I couldn't find what that is - the actual movements look well within the bounding box. I'll have to dig deeper.

kevinkahn, 2017-01-09:
Just FYI I have seen this as well on a FFCP with model sliced by Simplify3D.  Interestingly it only seemed to happen on 2 fairly thin models.  A taller (Z axis) model didn't issue the warning whereas a model that was about 1mm high did and another about 3-4mm high did.

foosel, 2017-01-10:
Found it. The culprit was a combination of a relative movement mode and a non-move `G0` or `G1` (e.g. to retract at the end of the print) causing relative position calculation to do utterly wrong stuff and duplicating coordinate values. This should now be fixed by the above commit. Is already pushed to the `maintenance` branch and will be part of the 1.3.1 release. Will also be merged into `devel` later.

SparseTheory, 2017-01-11:
Thank you!!!

- Jimmy
(Please excuse any misspellings or typos.)

closed 2017-01-25 by foosel

fform, 2017-01-31:
@foosel  I'm seeing this error with 1.3.1 (hadn't seen it prior with 1.2.16) using a Makerbot Replicator 2X using gcode from Simplify3d and the GPX plugin. Should I make a new issue or attach my gcode / logs here?

kevinkahn, 2017-01-31:
I'm still seeing it also on the FFCP after upgrading to 1.3.1 FYI

foosel, 2017-01-31:
@fform @kevinkahn example gcode files and profile settings for reproduction please, no way to determine if you are actually seeing the same issue or something else without more information than "me too"

PS: also might take a bit to check once information is there, currently down sick

kevinkahn, 2017-01-31:
Sure - of course you need an example.  Sorry I didn't initially send one.  Here's a gcode file that caused the issue with my FFCP.  It was sliced by Simplify3D if that matters and I can send the stl if you need it as well.  No rush on this at all since it doesn't really impact in any way given you can print anyway.  Just figured you'd want to know the behavior was still occurring.  Thanks.

[bezelv4top.zip](https://github.com/foosel/OctoPrint/files/742586/bezelv4top.zip)

SparseTheory, 2017-02-01:
All,

I recently upgraded to 6.3.1.  While I have not encountered the issue on
any new prints since upgrading, I do still get the issue if I load a
previous print (g-code) file under the files section.  Not sure if this is
expected behavior or not.

Regards,

Jimmhy

fform, 2017-02-16:
@foosel Here's a gcode file that gives me this error but prints fine
https://gist.github.com/fform/908a7c448a307f5a81ec4952650f8531

It's a Makerbot Rep 2X so the platform is 246mm x 152 mm x 154mm (xyz) and says that in the printer profile settings. This error only started with 1.3.1 (I came from 1.2.16). Even the same gcode files that I used on the old version show it and didn't before.

Cyphen-Cyn, 2017-07-29:
[Bed_Level_Calibration.txt](https://github.com/foosel/OctoPrint/files/1185290/Bed_Level_Calibration.txt)
i have the same issue here is the gcode provided below. The printer is the Replicator 1 dual.

Kunsi, 2017-07-30:
Please only report bugs related to latest release of OctoPrint. 1.3.1 is outdated, and 6.3.1 isn't even a valid verion number.

Cyphen-Cyn, 2017-07-30:
ill post a new one, not sure where you got 6.3.1 since i never posted a version number.

foosel, 2017-08-21:
@Cyphen-Cyn 

> since i never posted a version number

Indeed you didn't. You should have though ;) So what version are you seeing this on? Also please provide your printer profile information (bed width, depth, height & if configured custom bounding box values).

commit 6e474d90 (2017-01-10): Fix model size calculation during GCODE analysis

  * Properly handle G0/G1 with no X, Y, Z coordinates in relative mode
    instead of duplicating coordinates - should fix #1675
  * Only take move commands with X, Y, Z coordinates into account for
    model size calculation - this makes our internal GCODE analysis behave
    like the GCODE viewer's analysis and produce the same model size. The
    downside is that extrusions on the origin are no longer taken into account
    for checking if a model is within bounds of the print bed, but that should
    hopefully not produce any issues in the  [... cut]