OctoPrint/OctoPrint #1676: Incorrect line numbers after reset

camnewnham, 2016-12-29:
#### What were you doing?

Resetting the printer while idle or during single motions (not while printing), using the hardware button/LCD.

#### What did you expect to happen?

The printer would reset, and OctoPrint would not issue any immediate commands.

#### What happened instead?

Line numbers were not reset in OctoPrint, so the printer was requesting the incorrect data (ie.. OctoPrint was sending whatever it was previously at, eg. 59, but the printer had just been reset and hence was requesting line 1.) This occasionally meant that a sequence of incorrect commands were executed.

#### Branch & Commit or Version of OctoPrint

Version 1.3.0 (master branch)

#### Printer model & used firmware incl. version

Custom corexy. Repetier 1.0.0dev.

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

Terminal 1 - requesting line that is no longer in the buffer
https://gist.github.com/camnewnham/889486bfa8cf309ed1ebfa145106ef69
Terminal 2 - requesting and executing non-current commands after reset!!
https://gist.github.com/camnewnham/e826d644b40e794c9f240dac5868e9be

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A - however testing different values for 'Max Consecutive Timeouts While Idle" had no effect..

I have read the FAQ.

nophead, 2016-12-29:
Yes it all worked a lot better when immediate commands were sent without line numbers.

foosel, 2017-01-09:
@nophead I added an option explicitly for you to disable line numbers altogether ;) But they are necessary for a lot of printers out there to work reliably.

@camnewnham I'll look into it. It should hopefully suffice to simply send a reset line number command when a `start` is encountered during regular operation, since that usually indicates that the controller was reset.

nophead, 2017-01-09:
@foosel, Thanks for that, but where is it? I can't find such an option in Version: 1.3.0 (master branch).

foosel, 2017-01-09:
@nophead you need to disable firmware detection and then set "Send a checksum with the command" to "Never":

[image]

That should do it hopefully.

nophead, 2017-01-09:
Got it, thanks very much.

foosel, 2017-03-09:
OctoPrint will now reset line numbers via an `M110` when it encounters a `start` (exact match) while already operational. This should help in cases like the above. Needs some more real world tests though.

Change is pushed to `maintenance` and soon the `devel` branch and will be part of the 1.3.2 release. Some report back if it indeed helps with the problem described in this ticket would be appreciated.

foosel, 2017-03-16:
1.3.2 was just released.

closed 2017-03-16 by foosel