OctoPrint/OctoPrint #1679: [Request]  Support for Marlin AUTO_REPORT_TEMPERATURES

MoonshineSG, 2017-01-02:
Marlin has an advanced setting, that allows the firmware to auto report temperatures, at selected intervals, without the need of constant pull using M105. The availability of auto report can be found out via M115 when `EXTENDED_CAPABILITIES_REPORT` is turned on.

See more http://reprap.org/wiki/G-code#M155:_Automatically_send_temperatures

ntoff, 2017-01-04:
so apparently repetier firmware also supports that, +1 from me too.

fiveangle, 2017-06-05:
is this on the todo list ?

foosel, 2017-06-06:
@fiveangle no, it just got moved off the TODO list because the above commit implements it ;)

Already available on the `maintenance` branch, soon on `devel`, will be released with 1.3.5.

foosel, 2017-06-06:
As a sidenote though, current Marlin mainline still has both `EXTENDED_CAPABILITIES_REPORT` and `AUTO_REPORT_TEMPERATURES` disabled in all included example configurations, so unless you've intentionally enabled it, no dice. See also [my comments here](https://github.com/MarlinFirmware/Marlin/issues/5139#issuecomment-274097917).

I fear this also means that the next printer generation shipping with direct and unmaintained forks out of their vendor's factories will not be able to take advantage of either capability reporting or temperature autoreporting.

ntoff, 2017-06-06:
as a repetier firmware user, I welcome our new inferior marlin underlings.

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel