OctoPrint/OctoPrint #1682: [Request] Manual Mesh Bed Leveling via Octoprint Plugin?

pbrgle1, 2017-01-02:
Hi. It would be awesome if marlin's Manual Mesh Bed Leveling was also possible with a plugin and without an LCD and encoder. Not sure if it's that simple or even possible... Just throwing my idea out there :)

foosel, 2017-01-09:
That entirely depends on if the Marlin authors included some kind of GCODE commands to do whatever it is the LCD routine does at the moment - if so it could probably be done through [custom controls](http://docs.octoprint.org/en/master/features/custom_controls.html) already, if not there's nothing OctoPrint can do.

closed 2017-01-27 by foosel

fiveangle, 2017-05-31:
I found this looking for the same thing since it's a great idea.  Since the new bed levelling in Marlin 1.1 currently requires an LCD (which omits a large number of units) MBL will still be with us for a while, and doing MBL on a 5x5 grid through the terminal is inordinately tedious since super-fine-grained control is not available in OP by default.  Some printers like mine won't even move 0.01mm consistently so require an up/down movement to move that small increment with the most accuracy, exacerbating the problem.   With custom controls this looks more than viable and would vastly improve the process.  I'll see what I can do...

fiveangle, 2017-06-10:
Here you go.  Works like a charm !

[image]
[mbl_cust_controls.yaml.txt](https://github.com/foosel/OctoPrint/files/1065647/mbl_cust_controls.yaml.txt)

bertoo87, 2019-05-08:
> 
> 
> Here you go. Works like a charm !
> 
> [image]
> [mbl_cust_controls.yaml.txt](https://github.com/foosel/OctoPrint/files/1065647/mbl_cust_controls.yaml.txt)

could you tell me what to do with this file ? :o

Xennis, 2020-05-20:
> could you tell me what to do with this file ? 

For the sake of documentation: In the `config.yaml`. For details see https://docs.octoprint.org/en/master/features/custom_controls.html