OctoPrint/OctoPrint #1683: extrude button not taking steps into account

Foxandxss, 2017-01-03:
#### What were you doing?

Extruding 50mm of filament through the "Control" interface.

#### What did you expect to happen?

The extruding operation of having effect on the extruder's steps (Esteps in Marlin)

#### What happened instead?

It extrudes the same quantity of PLA with 90mm/s in Esteps or 300mm/s.

#### Branch & Commit or Version of OctoPrint

1.3.0

#### Printer model & used firmware incl. version

Anet A8, Marlin Rc7

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

to be added, got me printing already

#### Link to contents of terminal tab or serial.log

to be added, got me printing already, but it does a "G1 E50 F300"

#### Link to contents of Javascript console in the browser

No error to report

Other interfaces via direct usb (not using raspberry) say pronterface takes the steps in account so I could calibrate my extruder.

Foxandxss, 2017-01-03:
Not much else to be added, it is not a GUI issue nor I think the missing log will show anything useful.

nophead, 2017-01-03:
The log would show that OctoPrint sends the length in the box in a G1 E
command. If that doesn't change the length extruded there is something
wrong with your printer.

Foxandxss, 2017-01-03:
I can see a faulty printer, let's be honest, it is a piece of crap, but other software takes in account the steps.

Say I have 300 Esteps on the printer, if I extrude 50mm with octoprint (G1 E50 F300) it extrudes about 44mm. If I go to prontfrace and I click on extruding 50mm (also with the interface, not sending gcodes) it actually prints like 80mm (because 300 Esteps is like more than double of default).

So different apps have different results with the same command, prontface did allow me to calibrate my steps but with octoprint it always extrudes the same amount.

nophead, 2017-01-03:
How does OctoPrint know what E steps you have set in your firmware? It just
sends a length in mm and your firmware multiplies that by E steps and moves
the motor that number of steps.

Foxandxss, 2017-01-03:
That makes sense (sorry, I am not an expert with printers) but I was wondering why prontface was doing it correctly then.

If you feel that this is not an issue with octoprint, feel free to close, I will keep investigating.

foosel, 2017-01-09:
As @nophead already kindly pointed out, it's your firmware's job to translate `G1 E50 F300` into the needed step value. OctoPrint (or any host software for that matter) has no business knowing the steps-per-mm settings for the individual printer axes. The firmware already knows that, and it's its job to apply that knowledge correctly. Correspondingly, neither the `G0` nor the `G1` command do even allow to specify the movement in steps instead of mm, so there's nothing I even could change in that command to include steps.

I don't know why it works for you in pronterface, my guess would be it might be resetting your printer's eeprom values on connect for stuff to work correctly (although I do not know why that would not be persisted information). I suggest making sure your firmware knows the correct steps-per-mm settings for your printer axes, always, without having to fiddle with eeprom settings or anything like that - it's its job after all.

closed 2017-01-09 by foosel