OctoPrint/OctoPrint #1684: [Bug?] No unique file hashes

derpicknicker1, 2017-01-04:
#### What were you doing?

I sliced a *.stl file with the bundled cura plugin. I sliced it two times with same settings but different output file names.

#### What did you expect to happen?

I expect to get files with different file hashes

#### What happened instead?

All files have the same hash unless i use an other slicing profile.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0 (master branch)

#### Printer model & used firmware incl. version

Eustathios Printer with Marlin 1.0.2

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 55.0.2883.87 m (64-bit) on WIN10

#### Link to octoprint.log

https://gist.github.com/derpicknicker1/35d483e0f7da758ec57e13d433df96ac

#### Link to contents of terminal tab or serial.log

not needed

#### Link to contents of Javascript console in the browser

no output

#### Screenshot(s) showing the problem:

[image]

I have read the FAQ.

derpicknicker1, 2017-01-07:
I found the _create_hash function and it will generate the hash depending on the file data. So it is not a wonder that files with same slicing settings but different file names will have the same hash.
I think it will be a god thing, to inject the file name or file path into the hash. So the hash will become really unique and it will be possible to handle files by their hashes.
Here is my suggestion:
```
def _create_hash(self, path):
		import hashlib

		blocksize = 65536
		hash = hashlib.sha1()
		with open(path, "rb") as f:
			buffer = f.read(blocksize)
			while len(buffer) > 0:
				hash.update(buffer)
				buffer = f.read(blocksize)
                # insert the following Line:
                hash.update(path)
		return hash.hexdigest()
```

foosel, 2017-01-09:
That is actually how a file hash is supposed to work. The file name should not be relevant - if the contents are identical, the hash should be the same. So either I'm misunderstanding your issue here (I have to admit I'm still getting my head back into gear) or this is not a bug but behaving entirely as expected. If you slice the same file with the same profile, I'd expect the same content to be generated (unless there are placeholders in the file that are replaced with stuff like the current date), resulting in an identical hash. Which is the whole point since it allows reassociation (even though that currently isn't used).

derpicknicker1, 2017-01-09:
My problem is, that i use the hash for file handling. I thought that it will be unique. Using the path/file name is not a option for me, because i need to know the length, respectively the length should not exceed 40 characters.
This is for file browsing in Telegram plugin where i only can use 64 bytes of data in buttons.
So i think i have to find my own way. Maybe i generate hashes in a LUT.

foosel, 2017-01-09:
Ah, I see. Yeah, it's definitely not suited (and usable) for that. Could you maybe generate a new hash based on the path?

derpicknicker1, 2017-01-09:
Yes. That's the plan. As my question is answered now, i will close this issue. Thanks for your help.

closed 2017-01-09 by derpicknicker1