OctoPrint/OctoPrint #1685: [Bug?] Incomplete file analysis

derpicknicker1, 2017-01-04:
#### What were you doing?

I sliced a *.stl file with the bundled slicer plugin.

#### What did you expect to happen?

I expected that a complete file analysis will happen after slicing

#### What happened instead?

The metadata of the file will not include filament usage as they would when a analysis on uploaded gcode is done.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0 (master branch)

#### Printer model & used firmware incl. version

Eustathios Printer with Marlin 1.0.2

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 55.0.2883.87 m (64-bit) on WIN10

#### Link to octoprint.log

No output

#### Link to contents of terminal tab or serial.log

No output

#### Link to contents of Javascript console in the browser

No output

#### Screenshot(s) showing the problem:

Metadata of sliced file:

```
lego_deckel_test.gco:
  analysis:
    estimatedPrintTime: 5816
    filament:
      tool0: {}
      tool1: {}
  hash: 45698ee20bf37128dab8fbf5e499083a491852ee
  links:
  - hash: 666c9407805911c4aeb264fa57f24f98cb3d3451
    name: lego_deckel.stl
    rel: model
```

I have read the FAQ.

derpicknicker1, 2017-01-10:
After some investigation, i did a pull request to fix this issue.

foosel, 2017-01-11:
Fixed with your PR in `maintenance` and also `devel`, will be released with 1.3.1

foosel, 2017-01-25:
Part of 1.3.1 which has just been released.

closed 2017-01-25 by foosel