OctoPrint/OctoPrint #1686: Z Axis offset [Request]

ulmeck, 2017-01-04:
It would be very helpful for tweaking bed height adjustment if there was a way to set the z-axis offset in Octoprint control panel vs doing it in the slicer or firmware.

foosel, 2017-01-09:
If there's a GCODE command for that in your firmware (e.g. EEPROM setting), you can use [custom controls](http://docs.octoprint.org/en/master/features/custom_controls.html) for that. Otherwise there's nothing OctoPrint can do apart from rewriting all sent Z-coordinates, which might be a bad idea (and definitely only something supported through an external plugin).

ulmeck, 2017-01-09:
Noted. Was mostly looking for an LOE on it, which appears to be highish (on both dev and system req). I'll just create a UI element to control the EEPROM (Marlin) offset and be happy. :)

closed 2017-01-09 by ulmeck