OctoPrint/OctoPrint #1687: Login/Logout popup showed twice

gege2b, 2017-01-05:
Hi all and happy new year !
Well, this little bug is just an aesthetic one, it does not avoid octoprint to work properly

#### What were you doing?

Loging into my Octoprint server

#### What did you expect to happen?

The login successfull/failed notification to be showed only once

#### What happened instead?

It was showed twice
[image]

(in this screenshot, it's the failed login, but sucess login and logout is showed twice as well)

#### Branch & Commit or Version of OctoPrint

     Version: 1.4.0.dev22+gb3f35c5 (devel branch) 

#### Printer model & used firmware incl. version

not relevant

#### Browser and Version of Browser, Operating System running Browser

firefox 50.1.0 for ubuntu, on a fresh Kubuntu 16.10 x64 (installed yesterday)

#### Link to octoprint.log

```
2017-01-05 10:52:04,115 - tornado.application - ERROR - Uncaught exception GET /static/webassets/packed_libs.js?8f75e9e5 (192.168.1.222)
HTTPServerRequest(protocol='http', host='octoprint.local:5000', method='GET', uri='/static/webassets/packed_libs.js?8f75e9e5', version='HTTP/1.1', remote_ip='192.168.1.222', headers={'Accept-Language': 'en-US,en;q=0.5', 'Accept-Encoding': 'gzip, deflate', 'Host': 'octoprint.local:5000', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko/20100101 Firefox/50.0', 'Connection': 'keep-alive', 'Referer': 'http://octoprint.local:5000/', 'Pragma': 'no-cache', 'Cache-Control': 'no-cache'})
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/web.py", line 1334, in _execute
    result = yield result
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/gen.py", line 628, in run
    value = future.result()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/concurrent.py", line 109, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/gen.py", line 631, in run
    yielded = self.gen.throw(*sys.exc_info())
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/web.py", line 2172, in get
    yield self.flush()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/gen.py", line 628, in run
    value = future.result()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/concurrent.py", line 111, in result
    raise self._exception
StreamClosedError
2017-01-05 10:52:04,154 - tornado.general - ERROR - Cannot send error response after headers written
2017-01-05 10:52:04,277 - tornado.application - ERROR - Uncaught exception GET /static/webassets/packed_core.js?6e78dedd (192.168.1.222)
HTTPServerRequest(protocol='http', host='octoprint.local:5000', method='GET', uri='/static/webassets/packed_core.js?6e78dedd', version='HTTP/1.1', remote_ip='192.168.1.222', headers={'Accept-Language': 'en-US,en;q=0.5', 'Accept-Encoding': 'gzip, deflate', 'Host': 'octoprint.local:5000', 'Accept': '*/*', 'User-Agent': 'Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:50.0) Gecko/20100101 Firefox/50.0', 'Connection': 'keep-alive', 'Referer': 'http://octoprint.local:5000/', 'Pragma': 'no-cache', 'Cache-Control': 'no-cache'})
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/web.py", line 1334, in _execute
    result = yield result
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/gen.py", line 628, in run
    value = future.result()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/concurrent.py", line 109, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/gen.py", line 631, in run
    yielded = self.gen.throw(*sys.exc_info())
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/web.py", line 2172, in get
    yield self.flush()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/gen.py", line 628, in run
    value = future.result()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/concurrent.py", line 111, in result
    raise self._exception
StreamClosedError
2017-01-05 10:52:04,288 - tornado.general - ERROR - Cannot send error response after headers written
2017-01-05 10:52:11,636 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.222
2017-01-05 10:52:14,663 - octoprint.plugins.cura - INFO - Path to CuraEngine has not been configured yet or does not exist (currently set to None), Cura will not be selectable for slicing
2017-01-05 10:52:16,667 - tornado.access - WARNING - 401 POST /api/plugin/automaticshutdown (192.168.1.222) 97.24ms
2017-01-05 10:52:41,349 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 1.0s
2017-01-05 10:52:58,954 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-01-05 10:52:58,958 - octoprint.util.pip - INFO - Version of pip is 8.1.0
2017-01-05 10:53:25,918 - octoprint.util.pip - INFO - pip installs to /home/pi/OctoPrint/venv/lib/python2.7/site-packages/, --user flag needed => no, virtual env => yes
2017-01-05 10:53:26,044 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-01-05 10:55:02,562 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 63.54ms
2017-01-05 10:55:02,649 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 68.88ms
2017-01-05 10:55:15,776 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 62.94ms
2017-01-05 10:55:16,073 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 63.56ms
2017-01-05 10:55:20,169 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 66.93ms
2017-01-05 10:55:20,257 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 62.42ms
2017-01-05 10:55:27,426 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 62.90ms
2017-01-05 10:56:29,857 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.222
2017-01-05 10:57:11,803 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.222
2017-01-05 10:57:12,148 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.222
2017-01-05 11:07:45,540 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.222
```

#### Link to contents of terminal tab or serial.log

not relevant

#### Link to contents of Javascript console in the browser

Well, that's embarassing, 'cause right now, the problem is gone
The only thing I've done : mistyped my login a few times ( <= idiot) in a short lap of time (about 5 trys in 15/20 sec)
Each time, the popup showed twice, but since now, it seems to be OK.

BUT, because the problem was here and I already wrote all the text up to there, I'll post it anyway :)

#### Screenshot(s) showing the problem:

see above

I have read the FAQ.

foosel, 2017-01-09:
This part of your log:

```
2017-01-05 10:55:02,562 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 63.54ms
2017-01-05 10:55:02,649 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 68.88ms
2017-01-05 10:55:15,776 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 62.94ms
2017-01-05 10:55:16,073 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 63.56ms
2017-01-05 10:55:20,169 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 66.93ms
2017-01-05 10:55:20,257 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 62.42ms
2017-01-05 10:55:27,426 - tornado.access - WARNING - 401 POST /api/login (192.168.1.222) 62.90ms
```

suggests that you clicked on the Login button multiple times. Each time triggered an error on the backend due to wrong username/password. That in turn triggered an error message on the UI.

I guess it would make sense though to a) prevent multiple clicks from turning into multiple requests and b) preventing multiple concurrent popups from being generated here.

gege2b, 2017-01-09:
Yeah, as mentionned, I mistyped my login a few time when doing the test, but the double popup also showed twice before that.

It also showed twice this morning (and without any typo in my login, this time :)

The double popup seems to be  showed quite randomly...

foosel, 2017-01-09:
Weird. I've never seen it myself. The only explanation I have right now is multiple requests.

gege2b, 2017-01-09:
```
2017-01-09 09:45:36,243 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2017-01-09 09:45:36,273 - octoprint.plugin.core - INFO - Found 10 plugin(s) providing 9 mixin implementations, 6 hook handlers
2017-01-09 09:45:36,468 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2017-01-09 09:45:38,435 - octoprint.plugin.core - INFO - Found 10 plugin(s) providing 9 mixin implementations, 6 hook handlers
2017-01-09 09:45:38,467 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2017-01-09 09:45:47,130 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2017-01-09 09:45:47,429 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2017-01-09 09:45:56,438 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-01-09 09:46:04,803 - octoprint.util.pip - INFO - Version of pip is 8.1.0
2017-01-09 09:46:04,807 - octoprint.util.pip - INFO - pip installs to /home/pi/OctoPrint/venv/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-01-09 09:46:04,819 - octoprint.plugin.core - INFO - Initialized 9 plugin implementation(s)
2017-01-09 09:46:04,856 - octoprint.plugin.core - INFO - 10 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins/announcements
|  Automatic Shutdown (0.1.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_automaticshutdown
|  Core Wizard (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins/discovery
|  EEPROM Marlin Editor Plugin (0.1.0) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_eeprom_marlin
|  Plugin Manager (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins/softwareupdate
|  System Command Editor (0.3.1) = /home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_systemcommandeditor
|  Virtual Printer (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.4.0.dev22_gb3f35c5-py2.7.egg/octoprint/plugins/virtual_printer
2017-01-09 09:46:04,868 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-01-09 09:46:04,956 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2017-01-09 09:46:05,124 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2017-01-09 09:46:05,248 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-01-09 09:46:06,684 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octoprint for SSDP
2017-01-09 09:46:07,434 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2017-01-09 09:46:07,741 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2017-01-09 09:46:08,154 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.37s
2017-01-09 09:46:11,451 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://octoprint.local:5000/'}
2017-01-09 09:46:59,921 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.1.197:5000/'}
2017-01-09 11:21:19,419 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.222
2017-01-09 11:21:25,289 - tornado.access - WARNING - 401 POST /api/plugin/automaticshutdown (192.168.1.222) 80.85ms
2017-01-09 11:22:05,115 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-01-09 11:22:05,120 - octoprint.util.pip - INFO - Version of pip is 8.1.0
2017-01-09 11:22:31,208 - octoprint.util.pip - INFO - pip installs to /home/pi/OctoPrint/venv/lib/python2.7/site-packages/, --user flag needed => no, virtual env => yes
2017-01-09 11:22:31,331 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-01-09 11:22:32,157 - tornado.access - WARNING - 401 GET /api/files/local (192.168.1.222) 47.55ms
2017-01-09 11:22:41,488 - octoprint.filemanager.analysis - INFO - Starting analysis of local:20170109-112204-16_T90_0.2-25-.gcode
2017-01-09 11:22:41,515 - octoprint.printer.standard - INFO - Cannot load file: printer not connected or currently busy
```

Here is this morning log, the popup showed twice on the 11h21m19s login 
The second tornado access is due to Slic3r's "send to printer" feature

I will try to look for some pattern

BillyBlaze, 2017-01-09:
I am pretty sure this bug is caused by the code I did before the vacation. I fixed this by removing the onKeyUp event on the branch ```main/supportPasswordManagers```, @gege2b, could you verify?

@foosel;
It looks like I forgot to remove the [onKeyUp legacy code](https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/static/js/app/viewmodels/loginstate.js#L110), since we converted it to the form, the keyCode is unnecessary, even in this case triggering the form submit twice 😞 I corrected this in ce1344eb0ccf92fa79775e4b1e1efdca1883c62c

gege2b, 2017-01-09:
It's coherent, given that I didn't use my 3D printer for 2 weeks around christmas/new year,  the bug wasn't here until the update on Jan. 3rd

And the randomness of the double popup could be explained because sometimes I use the mouse, and sometimes the enter key to validate

I'll give a try tomorrow

gege2b, 2017-01-10:
I have not tried the SupportPasswordManagers branch (lack of time) , but I tried to validate with and without hitting Enter and indeed, the popup show twice only when I hit Enter.

So it seems you're right :)

foosel, 2017-01-12:
This is fixed on `maintenance` and also `devel` and was indeed a bug that never made it into a release.

closed 2017-01-12 by foosel