OctoPrint/OctoPrint #1690: [Brainstorming] Force absolute extrude values in filament estimate

Moredread, 2017-01-06:
The Craftware slicer generates an M83 (relative extrude values) at the start of the GCODE when a startup extrude is set, but generates absolute extrude values. This throws of the time and filament usage estimates, i.e. they are often **much** larger as expected, as `octoprint.util.gcodeInterpreter.gcode` uses M82/M83 to detect the extrusion value mode.

Would it be worthwhile to have a detection of Craftware GCODE to force the setting of absolute values? Or is there infrastructure for the user to replace text in the GCODE on the fly (per regex or string replacement) I haven't seen?

foosel, 2017-01-09:
Sounds like Craftware should fix their GCODE generator if I understand that correctly.

But in general, [the octoprint.filemanager.preprocessor hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-filemanager-preprocessor) might be of interest here - if there's a way to detect Craftware GCODE files easily, it could simply rewrite the contained `M83`, either by removing it completely or replacing it with `M82` instead or something.

nophead, 2017-01-09:
Seems legitimate GCODE to me. When using Cura I also switch between relative and absolute in my start.gcode and end.gcode because Cura only does absolute, which is no good for manual manipulation.

foosel, 2017-01-09:
`M83` followed by relative extrusion values I'd call legitimate. `M83` followed by absolute values sounds broken to me though.

nophead, 2017-01-09:
I assumed the OP meant that it did some relative extrusions in the start gcode and then switched to absolute. If there is no M82 in between then I can't see how i works. The firmware must be broken.

Of course, for the first extrude, relative and absolute are the same, so perhaps it just ignores M83.

foosel, 2017-01-09:
At least that is what I understood was the issue here, `M83`, no `M82` after but absolute extrusion values throughout the file anyhow. Maybe @Moredread could provide us with an example file though, to have some certainty of what is actually the case here? :)

I agree that I'd expect the printer to also go haywire in such cases, if it supports `M83` (who knows, maybe it doesn't).

Moredread, 2017-01-09:
@foosel is correct. It seems the Craftbot (the main target for Craftware) ignores M82/83, so I only noticed it when tracking down bogus filament usage values. I've attached a sample file below.

I'll look into using the hook @foosel mentioned above, the GCODE generated by Craftware should be detectable by the comment in the first line. Of course it'd be better if it is fixed in Craftware itself. :/

[craftbot cube.gcode.txt](https://github.com/foosel/OctoPrint/files/694915/craftbot.cube.gcode.txt)

I hope I haven't mixed up what absolute and relative values mean. Sorry in advance if this is the case.

foosel, 2017-03-08:
@Moredread sorry, I only now got around to take a look at the file. That GCODE definitely doesn't make sense, you didn't mix anything up, it in fact is generating something that doesn't match up.

A plugin utilizing the mentioned hook should be able to solve this. And yes, Craftware not producing nonsensical GCODE would be the actual solution, but I guess until that happens, someone writing such a plugin and publishing it on the repo would make some friends in the process (hint hint) :)

closed 2017-03-08 by foosel