OctoPrint/OctoPrint #1691: Webcam install and configure but no image

marseillai, 2017-01-06:
Hello.
I'm trying to make working this camera : http://www.banggood.com/No-Drive-Mini-USB-Camera-For-Raspberry-Pi-p-1023048.html

I've install octoprint. I can see the camera with lsusb : 
pi@octopi:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 5: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 5: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M

I have activate camera with raspi-config.
pi@octopi:~ $ ls mjpg-streamer/
_build             LICENSE                     output_http.so  TODO
cmake              Makefile                    output_rtsp.so  utils.c
CMakeLists.txt     mjpg_streamer               output_udp.so   utils.h
input_file.so      mjpg_streamer.c             plugins         www
input_http.so      mjpg-streamer-experimental  README.md
input_raspicam.so  mjpg_streamer.h             scripts
input_uvc.so       output_file.so              start.sh

but accessing 
http://127.0.0.1:8080/?action=snapshot
or
http://127.0.0.1:8080/?action=stream
never show me anything.
What do I miss ?

Thanks for your help.

marseillai, 2017-01-06:
Updated trying to comply rules. Sorry for the mistake.

Salandora, 2017-01-06:
This doesn't sound like a bug with octoprint but more like a problem with mjpeg_streamer and your camera.

Can you please post the output of "lsusb" and "systemctl status webcamd"
Maybe your webcam doesn't get recognized.
Did you tried it on a computer might it be a bad webcam?
Also include your octopi.txt please. Maybe there is something wrong. It would also be better to talk in IRC about this :-)

marseillai, 2017-01-07:
Hi salandora.
I must tell this is my first time with raspberry. And I didn't touch a linux since years now.... So I don't really know where is the problem.

Here is the requested output : 
pi@octopi:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 5: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
        |__ Port 5: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M

pi@octopi:~ $ systemctl status webcamd
● webcamd.service - LSB: webcam daemon
   Loaded: loaded (/etc/init.d/webcamd)
   Active: active (running) since Fri 2017-01-06 20:31:09 UTC; 10h ago
  Process: 436 ExecStart=/etc/init.d/webcamd start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/webcamd.service
           ├─465 /bin/bash /root/bin/webcamd
           └─551 ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -...

pi@octopi:~ $ cat /boot/octopi.txt (I removed some of the comment for the paste)
### Configure which camera to use
#camera="auto"
camera="usb"

### Additional options to supply to MJPG Streamer for the USB camera
#camera_usb_options="-r 640x480 -f 10"

### Additional webcam devices known to cause problems with -f
# Apparently there a some devices out there that with the current
# mjpg_streamer release do not support the -f parameter (for specifying
# the capturing framerate) and will just refuse to output an image if it
# is supplied.
# The webcam daemon will detect those devices by their USB Vendor and Product
# ID and remove the -f parameter from the options provided to mjpg_streamer.
# By default, this is done for the following devices:
#   Logitech C170 (046d:082b)
# Using the following option it is possible to add additional devices. If
# your webcam happens to show above symptoms, try determining your cam's
# vendor and product id via lsusb, activating the line below by removing # and
# adding it as shown examplatory.
# If this fixes your problem, please report it back so we can include the device
# out of the box.
#additional_brokenfps_usb_devices=("046d:082b" "aabb:ccdd")

### additional options to supply to MJPG Streamer for the RasPi Cam
# See https://github.com/foosel/OctoPrint/wiki/MJPG-Streamer-configuration
# for available options
# Defaults to 10fps
#camera_raspi_options="-fps 10"

Yes I tried the webcam on my computer and it worked fine.

I'll connect on IRC. Thanks.

marseillai, 2017-01-07:
Ok. Now it works fine.
I had to change  /boot/octopi.txt
and add
additional_brokenfps_usb_devices=("058f:5608")
Thanks.

closed 2017-01-07 by marseillai