OctoPrint/OctoPrint #1692: [Bug] calculate_solid_layer_count returns 1 more than expected

torch-senouen, 2017-01-06:
#### What were you doing?

While debugging a different issue I noticed that in the terminal inputs to Cura the variable downSkinCount=3. I thought this was strange because in the Cura slicing profile I created I set the layer_height: 0.2 and solid_layer_thickness: 0.4. I checked the function "calculate_solid_layer_count" in OctoPrint/src/octoprint/plugins/cura/profile.py by manual calculation. 

#### What did you expect to happen?
 I would expect that solid_layer_count=2 with the inputs layer_height: 0.2 and solid_layer_thickness: 0.4

#### What happened instead?
Since the function subtracts 0.0001 from layer_height before dividing solid_thickness by the total the answer becomes approximately 2.001. Then the function takes the ceiling of 2.001 which returns 3. There are two ways to change this to get the result I would expect: either change the ceiling to a floor or add 0.0001 to layer_height instead of subtracting.

#### Branch & Commit or Version of OctoPrint
Version: 1.3.0.post0.dev0+g7f5d03d (master branch)

#### Printer model & used firmware incl. version
MonoPrice Maker Select Plus with stock firmware.

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 55.0.2883.87 m (64-bit) on Windows 10 Pro Anniversary Update

#### Link to octoprint.log

http://pastebin.com/SkbqR6F2

#### Link to contents of terminal tab or serial.log
(putty term output)
http://pastebin.com/jTwzND1v

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

#### Cura Slicing Profile
http://pastebin.com/Ank8vXUg

I have read the FAQ. Yes, I have actually read the FAQ.

foosel, 2017-01-09:
Looks like that one was actually inherited from Cura back when, see Ultimaker/CuraEngine#140. Will look into it.

foosel, 2017-01-11:
Fixed on `maintenance` (and soon also `devel`), will be released with 1.3.1

commit 071c32ff (2017-01-11): Ported fix of solid layer count from Ultimaker/CuraEngine#140

Fixes #1692
closed 2017-01-25 by foosel