OctoPrint/OctoPrint #1693: profile.py does not parse perimeter_before_infill

torch-senouen, 2017-01-07:
#### What were you doing?

While trouble shooting print quality issues I wanted to experiment with printing the perimeter before infill (Cura option variable perimeterBeforeInfill). I went to check this option in my slicing profile and noticed the option was missing completely from my profile. Upon further investigation I noticed that OctoPrint/src/octoprint/plugins/cura/profile.py does not have a rule to parse "perimeterBeforeInfill" from "perimeter_before_infill" in the function convert_to_engine, or anywhere else in the program.

I found [Issue #1141](https://github.com/foosel/OctoPrint/issues/1141) that was closed because the ticket was incomplete.

As a temporary fix I have modified my own profile.py with a line in convert_to_engine to set "perimeterBeforeInfill": 1 which has fixed the problem.

#### What did you expect to happen?

I expected when setting the variable perimeter_before_infill: true in my Cura slicing profile that the CuraEngine option perimeterBeforeInfill would be set to true or 1.

#### What happened instead?

The option is not parsed and the default value false is used.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0.post0.dev0+g7f5d03d (master branch)

#### Printer model & used firmware incl. version
N/A

#### Browser and Version of Browser, Operating System running Browser
N/A

#### Link to octoprint.log

http://pastebin.com/SkbqR6F2

#### Link to contents of terminal tab or serial.log

http://pastebin.com/jTwzND1v

#### Link to Cura slicing profile

http://pastebin.com/Ank8vXUg

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2017-01-09:
Looks like that one [was added to Cura 15.04.5 which came out after initial implementation of OctoPrint's Cura plugin](https://github.com/daid/LegacyCura/commit/e8c05de1913a45d164ecd5417001fe068ff8c21e). I'll look into it.

foosel, 2017-01-11:
Fixed on `maintenance` (and soon also `devel`), will be released with 1.3.1

torch-senouen, 2017-01-11:
Great, thanks @foosel !

commit 7c74594f (2017-01-11): Adjusted Cura profile handling to match latest Cura Legacy version

  * Added/changed mappings of profile to engine settings to
    match Cura Legacy mapping:
    * perimeterBeforeInfill: taken from perimeter_before_infill (new,
      fixes #1693)
    * skinSpeed: taken from solidarea_speed (new)
    * raftAirGapLayer0: sum of raft_airgap and raft_airgap_all
    * raftAirGap: taken from raft_airgap_all (new)
    * raftFanSpeed: changed to 0
    * raftSurfaceThickness: taken from raft_surface_thickness (new)
    * raftSurfaceLinewidth & raftSurfaceLineSpacing: taken from
      raft_surface_line [... cut]
closed 2017-01-25 by foosel