OctoPrint/OctoPrint #1694: [BUG?] Deleting STL while printing leaves printer in unknown state

CarnufexG, 2017-01-07:
When deleting(accidently)  an STL while it is printing, the printer stops printing but leaves it in an uninitialized state; bed heater and hot end(s) ON. Perhaps, the "cancel print" functions should be run rather than just quiting and leaving everything in an unknown state.

derpicknicker1, 2017-01-08:
Do you really mean a *.stl file? I think this will happen if you delete a *.gco or *.gcode file while printing it.
But anyways. How do you delete a gcode file while printing it? This option is disabled in the Octoprint UI while printing.

CarnufexG, 2017-01-08:
Yes, I meant gcode file. And yes, I accidentally hit the trashcan icon and my print stopped. It looked like it thought it was printing. I tried to resume and refresh the UI but I ended up having to restart the print from the beginning. Version 1.3.0 (master branch).

Salandora, 2017-01-09:
That shouldn't happen at all...
Can you please try to reproduce this and upload the log files. 
Hopefully we get some more info on this.
But this sounds like a very specific race condition, you normally shouldn't be able to remove the file you're printing, can you also please specify where this file was located, e.g. root, folder, folder in folder, maybe a full path.

And also this is not a [REQUEST] but a [BUG?], so don't change the title just to make the bot happy!

foosel, 2017-01-09:
Fully filled out ticket template please. Also include JS console output. As @Salandora said, marking this as a request instead of providing the information for a *bug report* asked from you is a sure way to not get the issue resolved.

CarnufexG, 2017-01-09:
> <mention@noreply.github.com>
> leaves printer in unknown state (#1694)

Sorry, will do... away from machine(s) at the moment. I will cancel now, 
and redo request when I return from business trip.

foosel, 2017-01-11:
Could it be that the file was in a folder and you deleted *that* by any chance? Because I just fixed a bug in d83440d32d12974b685696c19c9e6a3709b5972c that allowed this to happen due to the file-in-folder-detection having an issue and not getting the full data it needed to work properly.

foosel, 2017-01-27:
Please request a reopen when you can provide the requested details (in case the issue even still persists).

closed 2017-01-27 by foosel

CarnufexG, 2017-01-27:
I will try to recreate the issue this weekend.

Sent: 1/27/2017 03:11:13