OctoPrint/OctoPrint #1695: UI does not load

nstephenh, 2017-01-08:
#### What were you doing?

Installed OctoPrint and attempted to load UI.

#### What did you expect to happen?
The interface to fully load.

#### What happened instead?
The interface did not fully load, as pictured in screenshot.

#### Branch & Commit or Version of OctoPrint

version 1.3 aka 7f5d03d0549bcbd26f40e7e4a3297ea5204fb1cc

#### Browser and Version of Browser, Operating System running Browser

Chrome 55. I also tried Firefox and Edge.

#### Link to octoprint.log
https://gist.github.com/nstephenh/ab35293e7bd6bf0d46eb6aaa1c6bc5e0

#### Contents of Javascript console in the browser
"Less has finished and no sheets were loaded. less.min.js:13"

#### Screenshot(s) showing the problem:

http://imgur.com/a/bP8cr

I have read the FAQ.

LeftyBC, 2017-01-08:
I'm also having this issue now on a brand new install of OctoPrint on 1.3.0 - ubuntu 16.10 - on git commit 7f5d03d0549bcbd26f40e7e4a3297ea5204fb1cc.  My other install of Octoprint from a week or so ago on the same version is fine.

```
2017-01-07 21:58:08,154 - octoprint.server - INFO - Starting OctoPrint 1.3.0 (master branch)
2017-01-07 21:58:08,251 - octoprint.plugin.core - INFO - Loading plugins from /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins, /home/colin/.octoprint/plugins and installed plugin packages...
2017-01-07 21:58:08,910 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2017-01-07 21:58:08,911 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-01-07 21:58:08,929 - octoprint.plugin.core - INFO - Loading plugins from /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins, /home/colin/.octoprint/plugins and installed plugin packages...
2017-01-07 21:58:09,031 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-01-07 21:58:09,035 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/colin/.octoprint/uploads...
2017-01-07 21:58:09,036 - octoprint.filemanager.storage - INFO - ... file metadata for /home/colin/.octoprint/uploads initialized successfully.
2017-01-07 21:58:10,097 - octoprint.util.pip - INFO - Using "/home/colin/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2017-01-07 21:58:11,139 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2017-01-07 21:58:11,139 - octoprint.util.pip - INFO - pip installs to /home/colin/OctoPrint/venv/lib/python2.7/site-packages, --user flag needed => no, virtual env => yes
2017-01-07 21:58:11,140 - octoprint.plugin.core - INFO - Initialized 7 plugin implementation(s)
2017-01-07 21:58:11,142 - octoprint.plugin.core - INFO - 7 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins/virtual_printer
2017-01-07 21:58:11,144 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2017-01-07 21:58:11,147 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-01-07 21:58:11,148 - octoprint.server - INFO - Reset webasset folder /home/colin/.octoprint/generated/webassets...
2017-01-07 21:58:11,149 - octoprint.server - INFO - Reset webasset folder /home/colin/.octoprint/generated/.webassets-cache...
2017-01-07 21:58:11,449 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on flashbot for SSDP
2017-01-07 21:58:11,565 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2017-01-07 21:58:11,638 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2017-01-07 21:58:11,854 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.21s
2017-01-07 21:58:29,405 - octoprint.server.util.flask.PreemptiveCache - INFO - Adding entry for / and {'query_string': 'l10n=en', 'path': u'/', '_count': 1, '_timestamp': 1483855109.405626, 'base_url': 'http://flashbot:5000/'}
2017-01-07 22:01:33,091 - octoprint.server - INFO - Starting OctoPrint 1.3.0 (master branch)
2017-01-07 22:01:33,411 - octoprint.plugin.core - INFO - Loading plugins from /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins, /home/colin/.octoprint/plugins and installed plugin packages...
2017-01-07 22:01:35,482 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2017-01-07 22:01:35,484 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-01-07 22:01:35,633 - octoprint.plugin.core - INFO - Loading plugins from /home/colin/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugins, /home/colin/.octoprint/plugins and installed plugin packages...
2017-01-07 22:01:35,736 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2017-01-07 22:01:35,738 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/colin/.octoprint/uploads...
2017-01-07 22:01:35,751 - octoprint.filemanager.storage - INFO - ... file metadata for /home/colin/.octoprint/uploads initialized successfully.
[... cut ...]
2017-01-07 22:02:42,337 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2017-01-07 22:02:42,371 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://flashbot:5000/'}
2017-01-07 22:03:47,602 - octoprint.server - INFO - Shutting down...
2017-01-07 22:03:48,278 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2017-01-07 22:03:48,278 - octoprint.events - INFO - Event loop shut down
2017-01-07 22:03:48,282 - octoprint.server - INFO - Goodbye!

```

LeftyBC, 2017-01-08:
Doesn't look like it's loading any CSS for some reason:

```html
<head><style type="text/css" id="alertifyCSS">.alertify-logs>*{padding:12px 24px;color:#fff;box-shadow:0 2px 5px 0 rgba(0,0,0,.2);border-radius:1px}.alertify-logs>*,.alertify-logs>.default{background:rgba(0,0,0,.8)}.alertify-logs>.error{background:rgba(244,67,54,.8)}.alertify-logs>.success{background:rgba(76,175,80,.9)}.alertify{position:fixed;background-color:rgba(0,0,0,.3);left:0;right:0;top:0;bottom:0;width:100%;height:100%;z-index:2}.alertify.hide{opacity:0;pointer-events:none}.alertify,.alertify.show{box-sizing:border-box;transition:all .33s cubic-bezier(.25,.8,.25,1)}.alertify,.alertify *{box-sizing:border-box}.alertify .dialog{padding:12px}.alertify .alert,.alertify .dialog{width:100%;margin:0 auto;position:relative;top:50%;-webkit-transform:translateY(-50%);transform:translateY(-50%)}.alertify .alert>*,.alertify .dialog>*{width:400px;max-width:95%;margin:0 auto;text-align:center;padding:12px;background:#fff;box-shadow:0 2px 4px -1px rgba(0,0,0,.14),0 4px 5px 0 rgba(0,0,0,.098),0 1px 10px 0 rgba(0,0,0,.084)}.alertify .alert .msg,.alertify .dialog .msg{padding:12px;margin-bottom:12px;margin:0;text-align:left}.alertify .alert input:not(.form-control),.alertify .dialog input:not(.form-control){margin-bottom:15px;width:100%;font-size:100%;padding:12px}.alertify .alert input:not(.form-control):focus,.alertify .dialog input:not(.form-control):focus{outline-offset:-2px}.alertify .alert nav,.alertify .dialog nav{text-align:right}.alertify .alert nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button),.alertify .dialog nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button){background:transparent;box-sizing:border-box;color:rgba(0,0,0,.87);position:relative;outline:0;border:0;display:inline-block;-webkit-align-items:center;-ms-flex-align:center;-ms-grid-row-align:center;align-items:center;padding:0 6px;margin:6px 8px;line-height:36px;min-height:36px;white-space:nowrap;min-width:88px;text-align:center;text-transform:uppercase;font-size:14px;text-decoration:none;cursor:pointer;border:1px solid transparent;border-radius:2px}.alertify .alert nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button):active,.alertify .alert nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button):hover,.alertify .dialog nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button):active,.alertify .dialog nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button):hover{background-color:rgba(0,0,0,.05)}.alertify .alert nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button):focus,.alertify .dialog nav button:not(.btn):not(.pure-button):not(.md-button):not(.mdl-button):focus{border:1px solid rgba(0,0,0,.1)}.alertify .alert nav button.btn,.alertify .dialog nav button.btn{margin:6px 4px}.alertify-logs{position:fixed;z-index:1}.alertify-logs.bottom,.alertify-logs:not(.top){bottom:16px}.alertify-logs.left,.alertify-logs:not(.right){left:16px}.alertify-logs.left>*,.alertify-logs:not(.right)>*{float:left;-webkit-transform:translateZ(0);transform:translateZ(0);height:auto}.alertify-logs.left>.show,.alertify-logs:not(.right)>.show{left:0}.alertify-logs.left>*,.alertify-logs.left>.hide,.alertify-logs:not(.right)>*,.alertify-logs:not(.right)>.hide{left:-110%}.alertify-logs.right{right:16px}.alertify-logs.right>*{float:right;-webkit-transform:translateZ(0);transform:translateZ(0)}.alertify-logs.right>.show{right:0;opacity:1}.alertify-logs.right>*,.alertify-logs.right>.hide{right:-110%;opacity:0}.alertify-logs.top{top:0}.alertify-logs>*{box-sizing:border-box;transition:all .4s cubic-bezier(.25,.8,.25,1);position:relative;clear:both;-webkit-backface-visibility:hidden;backface-visibility:hidden;-webkit-perspective:1000;perspective:1000;max-height:0;margin:0;padding:0;overflow:hidden;opacity:0;pointer-events:none}.alertify-logs>.show{margin-top:12px;opacity:1;max-height:1000px;padding:12px;pointer-events:auto}</style>
        <title data-bind="text: title">OctoPrint</title>

        <link rel="shortcut icon" href="/static/img/tentacle-32x32.png">
        <link rel="apple-touch-icon" sizes="114x114" href="/static/img/apple-touch-icon-114x114.png">
        <link rel="apple-touch-icon" sizes="144x144" href="/static/img/apple-touch-icon-144x144.png">

        <meta name="apple-mobile-web-app-capable" content="yes">
        <meta name="mobile-web-app-capable" content="yes">
        <meta name="apple-mobile-web-app-status-bar-style" content="black">

        <link href="" rel="stylesheet" media="screen">
<link href="" rel="stylesheet" media="screen">

<script src="/static/js/lib/less.min.js" type="text/javascript"></script>
        <script lang="javascript">
    var BASEURL = "/";
    var API_BASEURL = BASEURL + "api/";
    var PLUGIN_BASEURL = BASEURL + "plugin/";
    var GCODE_WORKER = "/static/gcodeviewer/js/Worker.js";

    var CONFIG_DEBUG =  false;

    var CONFIG_GCODEFILESPERPAGE = 5000;
    var CONFIG_TIMELAPSEFILESPERPAGE = 10;
    var CONFIG_LOGFILESPERPAGE = 10;
    var CONFIG_USERSPERPAGE = 10;
    var CONFIG_WEBCAM_STREAM = "None"; // deprecated, only left for compatibility reasons
    var CONFIG_ACCESS_CONTROL = true; 
    var CONFIG_ACCESS_CONTROL_ACTIVE =  false;
    var CONFIG_SD_SUPPORT = true; 
    var CONFIG_FIRST_RUN = true; 
    var CONFIG_TEMPERATURE_GRAPH = true; 
    var CONFIG_GCODE_SIZE_THRESHOLD = 20971520;
[... cut ...]
    var BRANCH = "maintenance";
    var LOCALE = "en";
    var AVAILABLE_LOCALES = {"de": {"display": "Deutsch", "english": "German", "language": "de"}, "en": {"display": "English", "english": "English", "language": "en"}};
    var SUPPORTED_EXTENSIONS = [".stl", ".gcode", ".gco", ".g"];

    var OCTOPRINT_VIEWMODELS = [];
    var ADDITIONAL_VIEWMODELS = [];
    var OCTOPRINT_ADDITIONAL_BINDINGS = [];

    
        
        
        var OCTOPRINT_INITIAL_TAB = "#temp";
    
</script>
    </head>
```

LeftyBC, 2017-01-08:
(Happens both on `maintenance` branch and `master` at 1.3.0 btw.)

javcz, 2017-01-08:
Same issue. The client scripts are completely missing:

[image]

javcz, 2017-01-08:
Strange, I have tested source from tag 1.2.9 with exactly the same result. No packed_lib.js nor packed_app.js mentioned in the result html. I'm new to the octoprint architecture, it may be a problem somewhere between the chair and keyboard in my case ...

javcz, 2017-01-08:
My last observation ... templates in octoprint/templates/javascripts.jinja2 are expanded and included, however the {{ ASSET_URL }} template is empty

pscrespo, 2017-01-08:
Same issue with new installation. Tested tags 1.3.0, 1.3.0rc2, 1.3.0rc1, and 1.2.18 (which was working before) with same results. Could it be some other dependencies that got updated recently?

JohnOCFII, 2017-01-08:
I didn't see operating system in your details above.  I'm on MacOS Sierra, and 1.3.0 is working fine in Chrome (v55) and Safari, as well as Mobile Safari on iOS.

nstephenh, 2017-01-08:
I tried on Ubuntu Gnome 16.10 and Windows 10, in Chrome on both, and Firefox on Ubuntu and edge on Windows 10. 

Last night I also tried 1.2.18 and got a similar, but different issue. A completely white page with this js log: https://gist.github.com/nstephenh/c6d7d5da0fbcc2a8c3655a04b2d6640c

javcz, 2017-01-08:
this fixes the problem:
https://github.com/miracle2k/webassets/commit/f11f7d300a539caead81fc28b01f760e6c3c9409

LeftyBC, 2017-01-08:
I was able to get the UI to mostly load (pops up the setup wizard dialog, but the dialog's steps are all mashed together and non-functional) by running `pip install --upgrade webassets` in my venv for Octoprint, which appears to have applied the fix mentioned by @javcz.  Getting closer!

pscrespo, 2017-01-08:
That command fixed the problem for me, but now I got stuck in the first run wizard (the "next" button doesn't work). Had to manually edit config.yaml with the following to bypass the wizard:
` server:`
`  firstRun: false`
`  seenWizards:`
`    corewizard: null`
`    cura: null`
`    softwareupdate: null`

javcz, 2017-01-08:
I have the same experience. Originally I was at 1.2.X and it worked there after webassets update (actually I only manually did the scope fix). After I have returned to 1.3.0 I also have the dialogbox issue. No errors in console, it seems the knockout does not bind the events.

murdocman, 2017-01-08:
@pscrespo how exactly did you get past the next button not working? im having the same problem and im not sure how you did it.

pscrespo, 2017-01-09:
Stop Octoprint and edit the file config.yaml on ~/.octoprint with nano or any text editor. You'll find a tag named "server: " with a "secretKey" value below. Paste the code I wrote under that value. Take a look at this: http://docs.octoprint.org/en/master/configuration/config_yaml.html

murdocman, 2017-01-09:
where exactly is the config file because i think i found it and did what you said and it still keeps going to the setup wizard

murdocman, 2017-01-09:
@pscrespo

pscrespo, 2017-01-09:
In Debian/Raspbian, it's in ~/.octoprint/config.yaml. What OS are you using? Here's a screenshot with how it should look: http://imgur.com/a/k45To

murdocman, 2017-01-09:
iam running this on windows 7

pscrespo, 2017-01-09:
Then it should be in %appdata%\OctoPrint\config.yaml. You can edit it with notepad. Mind the spaces on each line.

murdocman, 2017-01-09:
i went into the config file and it was already set to false so i added the other lines and it still keeps going to the wizard? any suggestions?

pscrespo, 2017-01-09:
Can you post a screenshot of the file? Or upload it?

murdocman, 2017-01-09:
also mine is located @ - %appdata%\Roaming\OctoPrint\ is that a problem?

murdocman, 2017-01-09:
[image]

murdocman, 2017-01-09:
@pscrespo do you want to see the entire file?

pscrespo, 2017-01-09:
You got me. As far as I know it should work. Maybe there are 2 missing spaces on "seenWizards:"?

murdocman, 2017-01-09:
missing spaces where?

nstephenh, 2017-01-09:
@LeftyBC 's fix worked for me. Thanks!

closed 2017-01-09 by nstephenh

LeftyBC, 2017-01-09:
I'll open a separate ticket for the setup wizard.

murdocman, 2017-01-09:
for my problem?