OctoPrint/OctoPrint #1698: Update 1.2.7 UpdateFix not working

shawnkirsch, 2017-01-10:
#### What were you doing?

Followed instructions via https://github.com/foosel/OctoPrint/releases/tag/1.2.8

Tried all possible

#### What did you expect to happen?

Update to 1.2.8

#### What happened instead?

Same, stuck on 1.2.7

#### Branch & Commit or Version of OctoPrint

1.2.7 Master Branch

#### Printer model & used firmware incl. version

Printrbot Play

#### Browser and Version of Browser, Operating System running Browser

Safari 10.12.2

#### Link to octoprint.log

https://gist.github.com/shawnkirsch/c5b96b7bd77e7a252074cb56f331aa01

#### Link to contents of terminal tab or serial.log

Not needed

shawnkirsch, 2017-01-10:
I have reread all the info.  I realize the wording in the manual switching of branches was worded weirdly.  I have updated some logs here:

https://gist.github.com/shawnkirsch/9bf7d58377e580aa3828db3589b96a50

shawnkirsch, 2017-01-10:
seriously a gitbot?!

foosel, 2017-01-10:
That's just a shot in the dark, but since in your second link it's complaining about not being able to clean up the install prior to install and the following setup call then fails too, please try [fixing your permissions](https://github.com/foosel/OctoPrint/wiki/FAQ#i-cannot-updateswitch-branches-i-get-something-about-permissions).

If that isn't it, a manual update might have to be performed.

> seriously a gitbot?!

Yes. Running after log files, version numbers and the like for nearly every ticket all the time becomes old fast, as does continuously answering the same questions already answered in the FAQ :)

closed 2017-01-24 by foosel