OctoPrint/OctoPrint #1699: Browser becomes unresponsive

jammi, 2017-01-10:
#### What were you doing?

Left the OctoPrint page open.

#### What did you expect to happen?

The page to stay responsive.

#### What happened instead?

Becomes unresponsive.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0.post0.dev0+g7f5d03d (master branch)

#### Printer model & used firmware incl. version

Wanhao Duplicator i3 v2.1, Repetier 0.9

#### Browser and Version of Browser, Operating System running Browser

Safari, 10.0.2 on macOS Sierra, 9.x on iPad iOS 9.

#### Link to octoprint.log

https://gist.github.com/jammi/8299c0855b88a6aa8e63ffc6314c6824

#### Link to contents of terminal tab or serial.log

Nothing unusual, just Repetier doing its `Recv: wait` thing

#### Link to contents of Javascript console in the browser
#### Screenshot(s) showing the problem:

Safari console, corresponding 404 for the sourcemaps when opening the console half an hour after the page was loaded and become unresponsive meanwhile, the rest in it is "normal":
http://i.imgur.com/y5Pi28V.png

I have read the FAQ.

foosel, 2017-01-10:
Infos from IRC:

  * websocket stays responsive longer then rest
  * buttons backed by a REST request stop working, REST request isn't even attempted
  * nothing logged in JS console or network tab
  * strictly local functions (filtering the terminal, opening/closing accordions and the like) continue to work

Currently unreproduced.

foosel, 2017-01-10:
Potentially related to #1523 ?

jammi, 2017-01-10:
Possibly yes, the only thing apart from the System menu and some local UI elements that became responsive was the Disconnect button in the Connnection accordion, and hitting Connect after that kicked it up. This was after several reloads of no socket activity, at which point I'd normally have restarted OctoPrint to get it going.

jammi, 2017-01-10:
While the OctoPrint window on the Mac had started showing just 0°C / 0°C after reloads, the page on the iPad still read some ambient room temperature from the printer, but nothing there was any more responsive than on the Mac.

foosel, 2017-01-10:
If the disconnect button works, it's not the communication with the backend that's the problem. Based on you saying that you "couldn't send commands" on IRC, I assume now that the problem rather lies with the communication to your printer breaking but the communication between OctoPrint's front-end and backend still working just fine?

I'm utterly confused now what the actual issue is you are experiencing. Please describe exactly, with as much detail as possible, what works, what doesn't, and what exactly makes it work again. I for one am currently completely in the unclear on what you are actually looking at ;)

foosel, 2017-01-10:
If you want, a screencast on YouTube showing the problem would also be very helpful here I think.

jammi, 2017-01-10:
Well, when nothing in the main view responds, seems like ui-only features are active and there's no indication of responsiveness and there are no errors, and you don't want to start a new print on a potentially unstable system, what can you conclude?

foosel, 2017-01-10:
> what can you conclude

Actually, you shouldn't conclude anything but simply describe exactly what you are seeing, as detailed and descriptive as possible ("Becomes unresponsive." is pretty non-descriptive and can mean a multitude of things), especially if you don't know how stuff fits together :)

That prevents misunderstandings on both sides as we had here - I honestly thought you were looking at network communication issues based on your descriptions here and on IRC. And preventing misunderstandings in turn saves valuable time otherwise lost on useless back-and-forth while trying to understand and ideally solve an issue (be it through a bug fix or a configuration change).

bolsoncerrado, 2017-01-20:
https://github.com/foosel/OctoPrint/issues/1523 ??

foosel, 2017-01-20:
@bolsoncerrado that's what I thought too, but after some more details @jammi actually appears to be running into issues where his printer stops responding to commands sent via serial, with the web frontend itself staying completely responsive.

jammi, 2017-01-21:
I later got UI hangs as well, where the printer was responsive and traffic was otherwise ongoing, but UI buttons wouldn't pass on, like terminal command line entry not being sent or cleared, control tab things not doing anything and so forth. The incident while opening this issue could've overlapped with such an event. Left-hand side panel worked, the right-hand side panels didn't. Disconnect-Connect didn't resolve, no number of reloads of the browser window on either device would work either. Restarting octoprint in this case was the only way to restore functionality.

bolsoncerrado, 2017-01-21:
Same boat then.

foosel, 2017-01-24:
When it happens the next time, please provide a screenshot of the contents of your Javascript Console, the Terminal output and `octoprint.log`. Ideally even a youtube video or other kind of screencast of what you are exactly seeing.

It's impossible to debug this based on the currently available information, especially since it sounds like there might be more than one issue occurring here now, one being printer communication, another being potentially a session issue or another problem with client-browser communication (against which however speaks that the "left side was working"). In short, no way to investigate or even have a clue as to what might be going on without more info - ideally a demonstration of things.

Most importantly though: not enough data/too ambiguous data for "same boat then" or other cases of "me too", at least if they don't come with the above additional information.

Please try to see this from my perspective - it's like someone called me to get their car fixed but all I so far know is that some time ago it made rattling noises on the right side, and later it made a squeaking sound on the left. I haven't seen the car or heard the noises so I'm completely at this dark at this point apart from "something might be wrong".

bolsoncerrado, 2017-01-24:
Well just happened today, a few minutes ago, that the Pi2 Printer that stands besides a Pi3 Printer had OctoPi FROZEN, ie no "lag" just total freezeing!

Although I had .log tuned off (damn!) I was able to record this:

https://www.youtube.com/watch?v=DLArAYGSJm0

In case it helps?

foosel, 2017-01-24:
Now that looks rather like the browser running TouchUI on your screen there crashed and that is nothing I can do anything about - I neither maintain the browser nor do I maintain the touch UI plugin.

bolsoncerrado, 2017-01-25:
You mean the browser crash crashed the whole Pi?
On other (random) scenarios, sometimes I do get "timeout" white screens from Chromium, and some other times I just get the lagggggg behaviour I mentioned. 

This "total crash" scenario is totally new to me, I just recorded it for documentation purposes as suggested.

foosel, 2017-01-25:
I meant that the browser crashed. If the whole Pi crashed, I'd usually (!) expect the display to shut off too, but there might be constellations where that wouldn't happen and everything just lock up.

In general, I know that a lot of people ignore that advice from me, but: I wrote OctoPrint explicitly in a way to keep load off the server by off-loading as much as possible to the client (running in the browser on another usually way more powerful system), so that the server running on e.g. a Pi would have enough resources to fully concentrate on the printing. Basically, OctoPrint was designed as a headless server running on low-cost and low-powered hardware, controlled *remotely* through a web based UI from a browser running on a *different* more high-powered piece of hardware.

Running a browser with the client UI on the same low powered system was actually never the intention, and hence might produce issues thanks to crashing browsers or simply lack of system resources or the like. I do not recommend that and never have. And yes, I know, a lot of people run it like that and it runs fine for them but still, I do not recommend it.

----

All in all though the above doesn't look (to me) what @jammi experienced, but it's hard to tell with the current amount of information.

bolsoncerrado, 2017-01-25:
gotcha. I'll upgrade to Pi3 and report back if nastyness keeps happening. as you say, if its a matter of resources power, pi3 and eventually pi4 will solve this issue alone.

closed 2017-04-13 by jammi

garryd123, 2018-03-08:
hey mine is doing the same thing even with octoprint with the raspberry pi3 and yet on the lap top it keeps working if i log in. The 7" raspberry screen just freezers idid a new install but did not help that was with the latest ver num14 but printer does not stop working i am not useing ui touch and no camera