OctoPrint/OctoPrint #1709: pip --upgrade upgrades to an incompatible release of Jinja

kontakt, 2017-01-13:
#### What were you doing?

Installing a .zip test build of my plugin with --upgrade to force a reinstall

#### What did you expect to happen?

Standard plugin installation and operation

#### What happened instead?

All settings pages were deformatted and placed on one page.

#### Branch & Commit or Version of OctoPrint

1.3.0 (Master Branch)

#### Browser and Version of Browser, Operating System running Browser

Firefox 50.1.0

#### Link to octoprint.log

http://pastebin.com/eFpMKxdv
(including plugin manager log due to relevance)

#### Link to contents of Javascript console in the browser

http://pastebin.com/Zfn357GQ

#### Screenshot(s) showing the problem:

https://gyazo.com/df456aca28b338f504293f6f7ac4bb04

I have read the FAQ. ✓

foosel, 2017-01-13:
Already fixed, duplicate of #1697. 

`master` received a push to solve this for people installing from source (resulting in 1.3.0post1). Fix will also be contained in 1.3.1. Plugin installs from within OctoPrint are intentionally done such that dependencies are not updated unless explicitly required (as in, --upgrade flag is only used in conjunction with the flag to stop an upgrade of dependencies), for exactly these kind of reasons (random dependency changes somewhere in the dependency tree causing such issues - sadly semantic versioning doesn't seem to be actually applied by most people causing backwards incompatibilities).

closed 2017-01-13 by foosel

kontakt, 2017-01-13:
My mistake, I searched "jinja" and saw no results.

foosel, 2017-01-13:
No harm done, thanks in any case! 

Just a very annoying problem ;)