OctoPrint/OctoPrint #1710: build borken for version g9650c71

sanchosk, 2017-01-14:
#### What were you doing?

I updated OctoPrint and tried to restart it.

#### What did you expect to happen?

I expected it to start and show new build version.

#### What happened instead?

I got error message:
`/home/octoprint/OctoPrint/venv/bin/octoprint: line 3: __requires__: command not found
import: unable to open X server `' @ error/import.c/ImportImageCommand/364.
from: can't read /var/mail/pkg_resources
/home/octoprint/OctoPrint/venv/bin/octoprint: line 8: syntax error near unexpected token `newline'
/home/octoprint/OctoPrint/venv/bin/octoprint: line 8: `    sys.exit('`

#### Branch & Commit or Version of OctoPrint

Branch devel, commit cf9e17e 4c971a9, version 1.3.1rc1-12-g9650c71

#### Printer model & used firmware incl. version

Cheap i3 clone

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

Nothing is written to the log.

#### Link to contents of terminal tab or serial.log

Nothing is written to the log.

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

sanchosk, 2017-01-14:
Issue no longer persists after restart :(

sanchosk, 2017-01-14:
So, I found the way how to reproduce.
When I start octoprint manually using 
`sudo su octoprint
~/OctoPrint/venv/bin/octoprint serve
`
it starts and works.
When I run it using service script (copy&paste from the how-to for raspberry pi in OctoPrint wiki), it will not start.
Trying to start the service manually using 
`sudo service octoprint start`
I end up with error
/home/octoprint/OctoPrint/venv/bin/octoprint: line 3: __requires__: command not found
import: unable to open X server `' @ error/import.c/ImportImageCommand/364.
from: can't read /var/mail/pkg_resources
/home/octoprint/OctoPrint/venv/bin/octoprint: line 8: syntax error near unexpected token `newline'
/home/octoprint/OctoPrint/venv/bin/octoprint: line 8: `    sys.exit('

Until now I was using the script without problem for ~month or so. It got broken after update to the commit in the issue report.

foosel, 2017-01-16:
Stock OctoPrint doesn't need an x server. I have no idea what it's trying to import there that does.

It's also complaining about a syntax error there, so maybe that is the actual issue. Could you please provide the contents of your /home/octoprint/OctoPrint/venv/bin/octoprint file? That file gets automatically generated by pip on install, but maybe something went wrong the which is causing these issues that's visible in the file.

Could you also provide some more information on where and how your OctoPrint instance is installed?

Generally, this does look very weird and more like a corruption issue than a bug. It's failing in a part I don't even control, in a very weird way.

sanchosk, 2017-01-16:
This might not be a problem of the octoprint file as, when started manually, it works OK.
However, this is the content:
`#!/home/octoprint/OctoPrint/venv/bin/python`
`# EASY-INSTALL-ENTRY-SCRIPT: 'OctoPrint==1.4.0.dev72-g9650c71','console_scripts','octoprint'`
`__requires__ = 'OctoPrint==1.4.0.dev72-g9650c71'`
`import sys`
`from pkg_resources import load_entry_point`
``
``if __name__ == '__main__':`
`    sys.exit(`
`        load_entry_point('OctoPrint==1.4.0.dev72-g9650c71', 'console_scripts', 'octoprint')()`
`    )`
``
It must be something weird with the environment.
To describe it:
- ARM-based MK802II setup
- Debian 8.7 jessie, regularly updated
- Python 2.7.9 installed using apt-get
- running on read-only filesystem with NFS-mounted writable directory for .octoprint
- OctoPrint built from github, development branch

The startup script for init.d is copy+pasted from the octoprint scripts (both init and default) with no modification (other than changing the user pi to user octoprint).
This is an experimentation platform, so I am able to rebuild it from scratch anytime if needed.

sanchosk, 2017-01-16:
Well, now, when checking details, I noticed there is a diff between the new octoprint.init and the one I copied into the init.d folder. Will try to replace the file and retest. The defaults look OK...

sanchosk, 2017-01-16:
That fixed the issue!
Comparing the init file I found out the old one ignored every argument passed :)
The fix from -z to -n made the trick apparently.
I wonder how is it possible this even worked at all until now.

closed 2017-01-16 by sanchosk

foosel, 2017-01-16:
I'm not entirely sure how an error in the calling script (the init script) can cause an import error in the generated executable O_O But if it works for you know, that's fine for me, especially since I had no idea what could be acting up there anyhow so far ;)

sanchosk, 2017-01-16:
I guess the problem was the OS update.
As the version of the build (in the venv) uses it's own python and libs, it worked until I updated the OS.
This coincided with the build of newer version of octoprint.
So as the libs of python changed, the ones in venv no longer matched and therefore the build sw did not work.
Thanks to the wrong condition for BASEDIR and CONFIGFILE these variables were simply ignored until now :(
But this is just my guess...

sanchosk, 2017-01-16:
Sorry, I could not let it go...
I could not understand how it could work if the parameters were ignored.
The problem is, that both the basedir and the configfile variables are commented out in my config.
But I found the issue.
The problem is that the old init script started the daemon using 
'start-stop-daemon --start --background --pidfile /tmp/octo.pid --make-pidfile --exec /home/octoprint/OctoPrint/venv/bin/octoprint --chuid octoprint --user octoprint --umask 022 --nicelevel=-2 -- --basedir --config serve --port=5000'
As both variables basedir and configfile are empty, the word "serve" is used as the name of the config file apparently.
The new init file parses the variables OK and realizes there is no configfile nor basedir definition, therefore the start-stop-daemon command looks like this:
'start-stop-daemon --start --background --pidfile /tmp/octo.pid --make-pidfile --exec /home/octoprint/OctoPrint/venv/bin/octoprint --chuid octoprint --user octoprint --umask 022 --nicelevel=-2 -- serve --port=5000'
And that's all the difference - using this command, OctoPrint starts OK :)
So, thanks again for your great work and keep this somewhere for people that will start to have problem with service autostart ;)