OctoPrint/OctoPrint #1713: issue when uploading .gcode file to SD

mickgalt, 2017-01-15:
#### What were you doing?

When trying to upload a file to the SD from the web interface to my Prusa i3 MK2 the progress bar shows the file is sending and when it's completed. 
The screen on the Prusa shows the name of the being uploaded bu the name of the file does not disappear from the screen when octoprint says the file has fully uploaded. 
if i then try and upload another file, nothing happens, octoprint doesn't now show the progress of the new file and the printer doesn't show the new file name on the screen.

On the printer, when i search for the first file that uploaded, it does not show it straight away. when i scroll down the list it shows a blank space where the file should be and then i scroll back up the name of the file shows (this might be a prusa i3 issue)
[video to show what happens](https://1drv.ms/v/s!AtMmOPhofyCWgbtPs8Vxaww7yQjl3A)

#### Branch & Commit or Version of OctoPrint

1.3.0

#### Printer model & used firmware incl. version

Prusa i3 Mk2 
Firmware 3.0.9

#### Browser and Version of Browser, Operating System running Browser

iexplorer 11
windows 7

#### Link to octoprint.log

http://pastebin.com/QzTtAYg0

Salandora, 2017-01-16:
Hello,

can you please upload a full serial.log file from a GCode file transfer.
This could also be a firmware bug, to determine if it is OctoPrint's fault or the firmwares fault we
need the serial.log.

closed 2017-01-30 by GitIssueBot

mickgalt, 2017-03-11:
hello, I have attached the serial log and octoprint log. I have updated firmware on printer and updated octopi and am still having the same problem.

#### What were you doing?

Sending a gcode file from PC to printer (mk2 prusa) via octoprint. pc is running win7 with IE11

#### What did you expect to happen?
when file sent complete the file would show up on the printer (mk2 prusa)

#### What happened instead?

file showed on on the printer but the file name was incorrect, it had an arrow symbol in the middle of the name and only had the extension .gco not .gcode.. the file could not be read by the printer  (mk2 prusa)

#### Branch & Commit or Version of OctoPrint

1.3.1 (master branch)

#### Printer model & used firmware incl. version

 mk2 prusa 3.0.10 firmware

#### Browser and Version of Browser, Operating System running Browser

windows 7 with IE11

#### Link to octoprint.log

https://gist.github.com/mickgalt/62e48f3d6c7e4d38701f20189b732655

#### Link to contents of terminal tab or serial.log

https://gist.github.com/mickgalt/d4c50257cdfcd055140ab03359c0a1d7

I have read the FAQ.

foosel, 2017-03-13:
From your serial log:

```
2017-03-11 13:02:58,565 - Send: N0 M110 N0*125
2017-03-11 13:02:59,467 - Recv: start
[...]
2017-03-11 13:03:36,279 - Recv: Invalid M code.
2017-03-11 13:03:36,281 - Recv: ok
```

Your printer apparently doesn't know what to do with the "reset line number" command `M110`, which I've never seen before and most certainly not with Mk2 firmware. It's basically one of the most fundamental commands.

```
2017-03-11 13:03:37,106 - Recv: Begin file list
2017-03-11 13:03:37,260 - Recv: echo:2������J���$��t�T���fZ'�p|�	�'	�Wi��,4����^C3V��{ez�W��b��?���R��v�����-iO���������|���n�J������]��n���k����=�)_�8�����g����<���s������sf��/���q#�����]�:v���hw��w�?�Q��������������E��65%��������u��1���B]�c��l����������^��xw��P�&���'�f�om�?u��L�Y����"�O���\d�%z����|��z�V}On�Ko�/z2_�]��u����� �{�C����!�����z���\y8��fY�<�Z����Ks��
��[�~��^��~��9�u��}������kK�����k=�y{GZ��2�,~��%��v�n%��:�h�����~����]���G��s���o~U��k��C�P3���=O�C�H�?�V6y���Jg���U��>m>�MdU�]���q���,�{�0~,���3��~��������Z����q_�#���~w���uy�ob?���]A��BM��:������8����{ad��R�N�kC��Y���:c����-�������]�����u�&vn�<�]���_������Y�a��~����Y���������~���N�d�)��?eJ����?��N��=N��{f���s��m��Uq�i��1z,���)k�����<_��k���2��}��=v�7r�-�f����������|�h�&�OgX���l}'N�5���`&����?��������QN��B=������f�}-
2017-03-11 13:03:37,325 - Recv: �W����g��������������������u_w�����O}�T��C}?S�}�Y��~�yNuZ�w>�?5;j����m��w��.9����[�z�<�"�m^��r�����o�����m':���w�o�{��'����q�st�������^��w���wnM���H�O;7���"���_n����}�B���t��3�z�}�-�|�-�7]�����o�S�'m�~�z+��+"X��o���H;�k4��;�Xv���z�*���mW�j��������w2������nr��������w�&���\g���!J������U�Gu���{=R�_������������h������;��~�?����t�n�V����������~�~�z����:��2�]�ez�-�_�H2zt/�m\���q����E��dW���������>����������k����v�W��/���]�g��z���_����.T;V�����O�������|��9.�����SQ�����Q�LxN|K�mwN4��;�>��1j���P����;����|��+���������]�����s���o�����[����7���j{�~�������5.k��5����������S������;_���!����N�u�M3�?��,%;��7��������H�������k������z������������\t����H�����u���������n�z���U�_�?��?�.����������z�"�����'�����b<����_��J�����K~?��	���
2017-03-11 13:03:37,330 - Recv: 1/
2017-03-11 13:03:37,454 - Recv: End file list
```

Your printer is returning binary garbage when OctoPrint tells it to return the list of files on your SD card, and also one single folder with nothing in it apparently.

```
2017-03-11 13:03:37,459 - Send: M28 /v2cali~1.gco
[...]
2017-03-11 13:03:37,463 - Recv: ok
2017-03-11 13:03:37,471 - Recv: echo:Now fresh file: /v2cali~1.gco
2017-03-11 13:03:37,509 - Recv: Writing to file: /v2cali~1.gco
```

Your printer says "all's well" after OctoPrint tells it to open a file for writing.

```
2017-03-11 13:03:37,517 - Send: N4 M107*33
2017-03-11 13:03:37,959 - Recv: Error:error writing to file
2017-03-11 13:03:37,961 - Recv: ok
```

While it claimed it opened the file for writing earlier, it falls on its face on every line OctoPrint sends to it while trying to actually write it to the file.

```
2017-03-11 13:04:30,472 - Recv: Error:error writing to file
2017-03-11 13:04:30,474 - Recv: ok
2017-03-11 13:04:31,843 - Recv: start
2017-03-11 13:04:31,851 - Recv: echo:echo: Last Updated: Feb 14 2017 15:35:19 | Author: (none, default config)
2017-03-11 13:04:31,855 - Recv: Compiled: Feb 14 2017
2017-03-11 13:04:31,860 - Recv: echo: Free Memory: 2670  PlannerBufferBytes: 1280
2017-03-11 13:04:31,864 - Recv: echo:Hardcoded Default Settings Loaded
2017-03-11 13:04:33,253 - Recv: echo:SD card ok
```

And then while still "writing" to the file, your printer suddenly restarts - it's unclear if that's due to its own decision or because you reset it.

```
2017-03-11 13:04:44,497 - Send: M105
2017-03-11 13:04:44,508 - Recv: ok T:25.5 /0.0 B:25.9 /0.0 T0:25.5 /0.0 @:0 B@:0
2017-03-11 13:04:47,380 - Recv: start
2017-03-11 13:04:47,391 - Recv: echo:echo: Last Updated: Feb 14 2017 15:35:19 | Author: (none, default config)
2017-03-11 13:04:47,395 - Recv: Compiled: Feb 14 2017
2017-03-11 13:04:47,401 - Recv: echo: Free Memory: 2670  PlannerBufferBytes: 1280
2017-03-11 13:04:47,406 - Recv: echo:Hardcoded Default Settings Loaded
2017-03-11 13:04:48,793 - Recv: echo:SD card ok
2017-03-11 13:04:49,498 - Send: M105
2017-03-11 13:04:54,499 - Send: M105
2017-03-11 13:04:59,500 - Send: M105
2017-03-11 13:05:04,502 - Send: M105
2017-03-11 13:05:09,509 - Send: M105
2017-03-11 13:05:14,519 - Send: M105
2017-03-11 13:05:19,527 - Send: M105
2017-03-11 13:05:24,528 - Send: M105
2017-03-11 13:05:29,529 - Send: M105
2017-03-11 13:05:34,537 - Send: M105
2017-03-11 13:05:48,859 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2017-03-11 13:05:48,861 - Send: M105
2017-03-11 13:06:18,881 - No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2017-03-11 13:06:18,887 - Changing monitoring state from 'Operational' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
2017-03-11 13:06:18,899 - Connection closed, closing down monitor
```

Then it restarts once more but never responds to sent commands anymore, which makes OctoPrint give up after a while and disconnect because there is simply no way to talk to it anymore.

[... cut ...] n though it claims it does or the SD card in your printer is broken. I'd be very surprised if it was the former, so try replacing the card. Whatever the core reason for this issue is though, it's something on your printer or your printer's firmware. OctoPrint is trying everything in its power to communicate with your printer to get a file transferred to it, but your printer isn't cooperating at all.

If a new SD doesn't fix it, I'd also suggest to get in touch with your printer vendor's support.

markaswift, 2017-04-30:
I have a similar issues with the MK2S, "Error:error writing to file".

foosel, 2017-05-02:
@markaswift not OctoPrint's issue, your printer is reporting an error -> check/swap its SD card, if that doesn't help ask your printer vendor's support