OctoPrint/OctoPrint #1715: Intermitten TypeError when starting a print

MennoLauwerens, 2017-01-15:
After uploading a file to the watch-folder and press print, I am sometimes getting this error:

```
Changing monitoring state from 'Printing' to 'Error: TypeError: 'unsupported operand type(s) for +=: 'NoneType' and 'unicode'' @ comm.py:startPrint:775'
Changing monitoring state from 'Error: TypeError: 'unsupported operand type(s) for +=: 'NoneType' and 'unicode'' @ comm.py:startPrint:775' to 'Offline: TypeError: 'unsupported operand type(s) for +=: 'NoneType' and 'unicode'' @ comm.py:startPrint:775'
Connection closed, closing down monitor
```

When this happens I have to restart Octoprint.

I'm using: Version: 1.3.0 (master branch) on a RaspberryPI

The printer is a prusa I3 style with Ramps and Repetier firmware.

Logfile: [Octoprint.log](https://gist.github.com/MennoLauwerens/941e6b1865358ba64dbcefc7fbad6b14#file-octoprint-log)

foosel, 2017-01-16:
That is a truncated `octoprint.log` you provided there. Please **never** truncate the log, always provide the full log. It contains important data to be able to understand how OctoPrint instance is set up (including version and plugin information) that is required to analyse issues like this.

Salandora, 2017-01-16:
Additionally this line looks interesting

`IOError: [Errno 2] No such file or directory: '/home/pi/.octoprint/watched/8f89f7e0-0f30-4e66-b04d-8323fb6b1aae.gcode'`

which states the file doesn't exist inside the watched folder, so maybe something while transfering to the watched folder went wrong.

closed 2017-01-30 by GitIssueBot

foosel, 2017-03-08:
As a side note, this was probably related to the same bug causing #1760 and/or #1765

bolsoncerrado, 2017-03-13:
I just had the same issue right now, on 1.3.1MB

Right after printing successfully a .gcode, temps were cooling down, I hit PRINT on another file in the list and got this:

--------
Send: M105
Recv: ok T:32.3 /0.0 B:32.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:32.2 /0.0 B:32.0 /0.0 @:0 B@:0
Send: M105
Recv: ok T:31.9 /0.0 B:32.1 /0.0 @:0 B@:0
Send: M105
Recv: ok T:31.9 /0.0 B:31.6 /0.0 @:0 B@:0
Send: M105
Changing monitoring state from 'Operational' to 'Printing'
Recv: ok T:31.6 /0.0 B:31.6 /0.0 @:0 B@:0
Send: N0 M110 N0*125
Recv: ok
See octoprint.log for details
Changing monitoring state from 'Printing' to 'Error: TypeError: 'unsupported operand type(s) for +=: 'NoneType' and 'unicode'' @ comm.py:startPrint:775'
Changing monitoring state from 'Error: TypeError: 'unsupported operand type(s) for +=: 'NoneType' and 'unicode'' @ comm.py:startPrint:775' to 'Offline: TypeError: 'unsupported operand type(s) for +=: 'NoneType' and 'unicode'' @ comm.py:startPrint:775'
Connection closed, closing down monitor
------

REstarted Octoprint, chose the same file, and now its printing fine. Just thought you'd like to know?

RistoNiinemets, 2017-03-31:
Something similar for me. Has happened for half a year almost now. When I connect the printer and start a print, it works. If I stop it (or it has finished) and upload new file and try to print it, the same error occours:

`The was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: TypeError: 'unsupported operand type(s) for +=: 'NoneType' and 'unicode'' @ comm.py:startPrint:780`

Log:
> 2017-03-31 06:20:22,177 - octoprint.util.comm - ERROR - Exception while processing line
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 2679, in getNext
>     line = self._handle.readline()
>   File "/home/pi/oprint/lib/python2.7/codecs.py", line 672, in readline
>     return self.reader.readline(size)
>   File "/home/pi/oprint/lib/python2.7/codecs.py", line 527, in readline
>     data = self.read(readsize, firstline=True)
>   File "/home/pi/oprint/lib/python2.7/codecs.py", line 486, in read
>     self.charbuffer += newchars
> TypeError: unsupported operand type(s) for +=: 'NoneType' and 'unicode'
> 2017-03-31 06:20:22,401 - octoprint.util.comm - ERROR - Error while trying to start printing
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 780, in startPrint
>     line = self._getNext()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 1823, in _getNext
>     line = self._currentFile.getNext()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 2690, in getNext
>     raise e
> TypeError: unsupported operand type(s) for +=: 'NoneType' and 'unicode'
> 2017-03-31 06:20:22,200 - octoprint.util.comm - ERROR - Caught an exception in the send loop
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 2115, in _send_loop
>     self._continue_sending()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 1529, in > _continue_sending
>     elif self._sendNext():
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 1853, in _sendNext
>     line = self._getNext()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 1823, in _getNext
>     line = self._currentFile.getNext()
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/util/comm.py", line 2661, in getNext
>     raise ValueError("File %s is not open for reading" % self._filename)
> ValueError: File /home/pi/.octoprint/uploads/3DBenchy_R002.gcode is not open for reading
> 2017-03-31 06:20:23,009 - octoprint.events - INFO - Executing a system command
> 2017-03-31 06:20:25,745 - octoprint.filemanager.analysis - INFO - Starting analysis of local:3DBenchy_R02.gcode

By the log file it seems that something is wrong with the file. So now I connect again, "Load" file, wait a few seconds, and hit "Print" and now it works.

foosel, 2017-03-31:
See #1838. You might want to test the `maintenance` branch to verify the fix mentioned in that ticket indeed solves the issue for you.