OctoPrint/OctoPrint #1717: UI during transfer shows as printing information

KairuByte, 2017-01-16:
#### What were you doing?
Uploading a print to the SD card.

#### What did you expect to happen?
State to show transfer information, labeled as transfer information. Cancel and pause appear to not function in this state as well.

#### What happened instead?
Transfer information is labeled as printing information, and includes non transfer related fields. Cancel and pause should work.

#### Branch & Commit or Version of OctoPrint
Version: 0+unknown
However to be accurate I cloned the latest master as of time of this writing.

#### Printer model & used firmware incl. version
N/A (UI Bug)

#### Browser and Version of Browser, Operating System running Browser
Chrome (Latest), Windows 10

#### Link to octoprint.log
N/A (UI Bug)

#### Link to contents of terminal tab or serial.log
N/A (UI Bug)

#### Link to contents of Javascript console in the browser
N/A (UI Bug)

#### Screenshot(s) showing the problem:
[image]

foosel, 2017-01-16:
That's not a bug actually but just how stuff has always been done because it's really just cosmetic in nature.

Pausing an SD transfer also doesn't make much sense (either you want the transfer to proceed as fast as possible, or you want it to stop, you also can't pause a copy file dialog), it should be reflected though that the Pause button is unavailable then. Cancelling it does make sense, so this should be implemented.

KairuByte, 2017-01-16:
No pausing makes perfect sense, not sure exactly what I was thinking to be honest.

I had actually written, and then deleted, that perhaps the status card isn't the best place for the transfer information to appear? I will admit to having hastily hitting cancel because I think I somehow hit print on something, not realizing that while my upload was complete, the transfer was not. Although the same effect may be possible if the print/pause buttons simply disappear as it would give a distinct visual contrast to print mode.

foosel, 2017-01-16:
From an implementation perspective, it's a print job (even though the printer doesn't execute the commands but instead saves them to a file, what OctoPrint does is pretty much identical). So the state is a good location, especially since it makes clear that stuff is currently busy and it's not possible to print a file right now. But I agree that a better reflection of things would be valuable. Maybe just muting the things that don't make sense (e.g. timelapse), removing the reference to "Prints" from the labels in general ("Approx. Total Time", "Time" or maybe "Job Time", "Time Left" or maybe "Job Time Left", "Done" or maybe "Sent") and maybe replacing the three buttons just with one giant cancel button.

> No pausing makes perfect sense, not sure exactly what I was thinking to be honest.

It's monday :)

KairuByte, 2017-01-16:
Those sound like perfect changes to ensure the user doesn't get confused. If I had more time and had spent more then a few minutes in python over the past few years I would totally make the changes and pull request. I honestly don't know how you spend so much time on this.
> It's monday :)
Oh boy, I missed the clock tick over. Today is going to be a fun day at work.

foosel, 2017-01-16:
> I honestly don't know how you spend so much time on this.

It's my full time job and it's been that way since August 2014, first thanks to a single sponsor, now [thanks to the community](https://patreon.com/foosel). Otherwise I'd definitely not be able to keep this pace :D

KairuByte, 2017-01-16:
Nice! I wish I could dive full time into something like that, but I would be too afraid of failure. I'm not at a point in my life that I can justify taking that gamble with the family's well being, but I hope to get there one day!

ntoff, 2017-01-17:
>you also can't pause a copy file dialog

*pssst*, you can in windows 10 ;)

foosel, 2017-01-17:
You can? O_o That... that makes no sense at all o_O

BillyBlaze, 2017-01-17:
Offtopic, but, Yes, we can 👍 

[image]

foosel, 2017-01-17:
I'm shocked, SHOCKED!

TTN-, 2017-01-20:
Yea its great when someone cuts 10+ movies at the same time in parallel from one disk to the other, to stop the disk thrashing just pause most and let it run sequentially :)