OctoPrint/OctoPrint #1718: Connecting to printer aborts SD print

KairuByte, 2017-01-18:
#### What were you doing?
Connected to printer with existing print originating from SD running.

#### What did you expect to happen?
Print to continue.

#### What happened instead?
Print aborted. (Not aborted by Octoprint mind you. It just halts)

#### Branch & Commit or Version of OctoPrint
1.3.0

#### Printer model & used firmware incl. version
Anet A8 running Skynet3D 2.3.1

#### Browser and Version of Browser, Operating System running Browser
Chrome on Windows 10

#### Link to octoprint.log
Quite short, hope you dont mind if I paste it here.
> 2017-01-18 03:23:02,506 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.2
> 2017-01-18 03:23:06,039 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.23s
> 2017-01-18 03:23:12,999 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
> 2017-01-18 04:18:00,579 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin SkyNet3D-v2.3.1 (Github)"

#### Link to contents of terminal tab or serial.log

> Send: M105
> Recv: ok T:23.8 /0.0 B:24.0 /0.0 @:0 B@:0

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s) showing the problem:
N/A

I have read the FAQ.

I tried searching for this in open issues but didn't find anything. My apologies if this is a) a duplicate and/or b) not a bug, and an expected part of connecting.

foosel, 2017-01-18:
It sadly is b), at least for a lot of controllers out there which simply reset on serial connect. There's nothing I can do about that, sorry. It's also not possible to show a warning about that since I don't know if the printer is printing from SD unless I connect to it. And always showing a warning on connect would be extraordinarily annoying.

closed 2017-01-18 by foosel

KairuByte, 2017-01-18:
Fair enough. At least it doesn’t connect automatically which would exacerbate the problem.