OctoPrint/OctoPrint #1720: Support for endoscope/dual cameras

bsculley, 2017-01-19:
I have recently purchased an endoscope-style USB camera for my Prusa Mk2.  It allows a close-up view of the print head to assist in insuring the print is proceeding normally and in diagnosing any issues.  

It would be great if OctoPrint had support for this camera, and even better if there was a method of switching views between the endoscope and the normal camera.

foosel, 2017-01-27:
From the [FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#my-webcam-is-not-working):

> OctoPrint doesn't directly control your webcam. All it can do is embed an MJPG stream into the web interface and retrieve snapshots for the timelapse frames from the configured snapshot URL. If your webcam isn't working, it isn't something OctoPrint is at fault for or can fix.

So there is nothing I can do in OctoPrint to make this camera of your's work, sorry.

The request for a second webcam is already tracked under #401.

closed 2017-01-27 by foosel

hamselvdk, 2017-11-22:
Has this been startet?