OctoPrint/OctoPrint #1722: Smoothieboard - allow upload of "config.txt" & "firmware.bin" to SD

pflugj, 2017-01-20:
First of all:  Thanks for this wonderful software! 

#### Branch & Commit or Version of OctoPrint
Version 1.3.1rc1 (rc/maintenance branch)

#### Printer model & used firmware incl. version
DIY printer & Smoothieboard
Firmware: Smoothieware - Edge from 2017-18

#### Browser and Version of Browser, Operating System running Browser
Firefox

#### Screenshot(s) showing the problem:
[image]

Could you please allow for uploading those 2 files to the SD of smoothieboard?
Smoothieboard can be updated by simply copying a new "firmware.bin" to the SD
and
the configuration can be changed by overriding the existing "config.txt" on the SD.

Thank and best regards from Namibia
Jurgen

foosel, 2017-01-20:
Duplicate of #1652.

closed 2017-01-20 by foosel

pflugj, 2017-01-21:
I did not find this during my search ;-(
...sorry for the duplicate