OctoPrint/OctoPrint #1723: softwareupdate doesn't find latest releases on OctoPi

eyal0, 2017-01-21:
#### What were you doing?

Trying to force updates to plugins.

#### What did you expect to happen?

Expected softwareupdate plugin to find the new versions

#### What happened instead?

The softwareupdate plugin didn't find the latest versions.

#### Branch & Commit or Version of OctoPrint

1.3.0

#### Printer model & used firmware incl. version

OctoPi

#### Browser and Version of Browser, Operating System running Browser

Chrome

I have read the FAQ.

eyal0, 2017-01-21:
I've alreday debugged this issue.  The problem is that the OctoPi default pkg_resources version that is installed is old and returns a tuple instead of a Version object.  This is from OctoPi's python:

```
pi@octopi:~ $ ./oprint/bin/python
Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pkg_resources
>>> pkg_resources.parse_version("v1.2.3")
('*v', '00000001', '00000002', '00000003', '*final')

```

This is from my desktop system:

```
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pkg_resources
>>> pkg_resources.parse_version("v1.2")
<Version('1.2')>
```

Above, we can see that the OctoPi python pkg_resources returns a tuple.

From `src/octoprint/plugins/softwareupdate/version_checks/github_release.py` line 130, the code that converts the old style tuple into a version for comparing is in `_get_base_from_version_tuple`:

```
        base_version = []
        for part in version_tuple:
                if part.startswith("*"):
                        break
                base_version.append(part)
        return tuple(base_version)

```

GitHub recommends that version tags start with a "v".  With the code above, all versions that start with a "v" will simply become an empty tuple.  This causes the version comparison code to fail.

eyal0, 2017-01-21:
The latest pkg_resources has a more complicated version algorithm in `pkg_resources/_vendor/version.py` .  It's probably too complicated to implement in OctoPrint but it allows for skipping an initial v in versions in the regular expression `VERSION_PATTERN`.

https://github.com/pypa/setuptools/blob/master/pkg_resources/_vendor/packaging/version.py#L160

To keep things, OctoPrint can simply ignore a leading v in the tuple.

foosel, 2017-01-24:
PR by @eyal0 to improve this behaviour has been backported to `maintenance` and will be released with 1.3.2

closed 2017-01-25 by foosel