OctoPrint/OctoPrint #1725: Octopi upgrade problem installing webassets

m0rgano, 2017-01-22:
Not sure if this is the right place to ask for problems with installation upgrade? If not please point me in right direction.

I just tried to upgrade my version of Octopi from rather old V1.1 to latest version.

I followed the manual upgrade procedure but I get an error when the script tries to install the webassets package:

### Processing OctoPrint-1.3.0.post1_g1014712-py2.7.egg
removing '/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg' (and everything under it)
creating /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg
Extracting OctoPrint-1.3.0.post1_g1014712-py2.7.egg to /home/pi/oprint/lib/python2.7/site-packages
OctoPrint 1.3.0.post1-g1014712 is already the active version in easy-install.pth
Installing octoprint script to /home/pi/oprint/bin

Installed /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.0.post1_g1014712-py2.7.egg
Processing dependencies for OctoPrint==1.3.0.post1-g1014712
Searching for webassets>=0.10
Reading http://pypi.python.org/simple/webassets/
Best match: webassets 0.12.1
Downloading https://pypi.python.org/packages/9a/35/d86c6d3bd31e2e9c8351c8766ef3688c75d56bc5233ed49061e522efc3cc/webassets-0.12.1.tar.gz#md5=429401824556e6e0868e437f9e561ba6
Processing webassets-0.12.1.tar.gz
Running webassets-0.12.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-N6ptRU/webassets-0.12.1/egg-dist-tmp-Yx1fyf
error: /tmp/easy_install-N6ptRU/webassets-0.12.1/src/webassets/__init__.py: No such file or directory

Any ideas how I can fix this? 

Thanks

m0rgano, 2017-01-22:
Should have also said now when I try to connect from a browser I get 'Error 503 Service Unavailable'. This was working fine before the attempted upgrade.

The octoprint.log file contains only one line indicating a connection attempt from my IP.

If you need any more information please ask - I am fairly fluent in Linux, just no real experience with Python.

foosel, 2017-01-24:
While the error points at there being an issue with the `webassets` dependency (which I wouldn't be able to fix in any case, I could only pin to a working version), I cannot reproduce this.

I did a test install freshly from source as well as an upgrade from 1.2.x without any issues:

```
Searching for webassets>=0.10
Reading https://pypi.python.org/simple/webassets/
Best match: webassets 0.12.1
Downloading https://pypi.python.org/packages/9a/35/d86c6d3bd31e2e9c8351c8766ef3688c75d56bc5233ed49061e522efc3cc/webassets-0.12.1.tar.gz#md5=429401824556e6e0868e437f9e561ba6
Processing webassets-0.12.1.tar.gz
Writing /tmp/easy_install-ZvsBAj/webassets-0.12.1/setup.cfg
Running webassets-0.12.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-ZvsBAj/webassets-0.12.1/egg-dist-tmp-i0oWJQ
warning: no files found matching 'run_tests.py'
no previously-included directories found matching 'docs/_build'
warning: no previously-included files matching 'out.css' found under directory 'examples'
warning: no previously-included files matching 'out.js' found under directory 'examples'
no previously-included directories found matching 'examples/appengine-sdk'
warning: no previously-included files matching '*.pyc' found anywhere in distribution
warning: no previously-included files matching '.gitignore' found anywhere in distribution
warning: no previously-included files matching '*.orig' found anywhere in distribution
warning: no previously-included files matching 'webassets-cache/*' found anywhere in distribution
warning: no previously-included files matching '.sass-cache/*' found anywhere in distribution
zip_safe flag not set; analyzing archive contents...
webassets.filter.__init__: module references __path__
Adding webassets 0.12.1 to easy-install.pth file
Installing webassets script to /home/pi/OctoPrintTest/venv/bin

Installed /home/pi/OctoPrintTest/venv/lib/python2.7/site-packages/webassets-0.12.1-py2.7.egg
```

I also did at least 10 upgrades and installs over the course of preparing the release candidates of 1.3.1 and didn't run into any issues then either. 

Can you provide more details on the version you are upgrading from, the version you are upgrading to and the exact commands you are using to upgrade? What python version is this (`/home/pi/oprint/bin/python --version`), what version of pip (`/home/pi/oprint/bin/pip --version`) and what version of setuptools (`/home/pi/oprint/bin/easy_install --version`)?

For reference:

```
(venv)pi@octopi:~/OctoPrintTest $ python --version
Python 2.7.9
(venv)pi@octopi:~/OctoPrintTest $ pip --version
pip 1.5.6 from /home/pi/OctoPrintTest/venv/local/lib/python2.7/site-packages (python 2.7)
(venv)pi@octopi:~/OctoPrintTest $ easy_install --version
setuptools 5.5.1
```

m0rgano, 2017-01-24:
Hi Gina,

Thanks for your reply.

pi@3DPrinterRaspi ~ $ python --version
Python 2.7.3
pi@3DPrinterRaspi ~ $ pip --version
pip 1.1 from /usr/lib/python2.7/dist-packages (python 2.7)
pi@3DPrinterRaspi ~ $ easy_install --version
distribute 0.6.24dev-r0

The previous version of OctoPrint was 1.1.x

I can't remember the rest of the version number was - can I find that 
anywhere now that I have attempted the upgrade?

The version I was trying to upgrade to was the latest stable version 
(done by doing the git pull). Again is there anywhere I can find that 
information on my Pi?

Regards,

Jeremy

foosel, 2017-01-24:
Your above upgrade log showed that you had used a virtualenv located at `~/oprint/` for installation. You'll need to provide the version numbers of python, pip and setuptools for that virtualenv - the above is from the global versions. Usually they are the same but let's make sure. Especially important since your version of `easy_install` is supplied by distribute instead of setuptools, which might be the clue here.

So please do this:

```
source ~/oprint/bin/activate
python --version
pip --version
setuptools --version
```

or use the exact commands I told your to execute above (NOT the ones from my output, which was generated for my test install that is different from yours).

Is this an OctoPi install? If so please also provide its version: `cat /etc/octopi_version`. The fact that there's `distribute` in the mix instead of `setuptools` there worries me, I'm fairly sure we always shipped setuptools with OctoPi.

Also provide a link to your `~/.octoprint/logs/octoprint.log` file (which was actually also requested from you in the bug template ;)). The last installed version should hopefully be logged in that.

Finally, just to make that clear, if the install you are upgrading there is as ancient as it sounds, it might be necessary (and at the very least easier) to just backup everything under ~/.octoprint and finally flash a fresh image. You are trying to upgrade from something there that's from 2014 or even older than that - it should in theory work via a git pull, but so many things have changed since then not only in OctoPrint but also in Raspbian, Python and so on that it's impossible to tell how such an old environment might influence things, especially considering the equally ancient python and pip versions on your system. I cannot guarantee anything here, sorry, too many variables.

m0rgano, 2017-01-24:
Hi Gina,

pi@3DPrinterRaspi ~ $ source ~/oprint/bin/activate
(oprint)pi@3DPrinterRaspi ~ $ python --version
Python 2.7.3
(oprint)pi@3DPrinterRaspi ~ $ pip --version
pip 1.1 from 
/home/pi/oprint/lib/python2.7/site-packages/pip-1.1-py2.7.egg (python 
2.7) (oprint)

pi@3DPrinterRaspi ~ $ setuptools --version
-bash: setuptools: command not found
(oprint)pi@3DPrinterRaspi ~ $

So as you can see, no setuptools.

As far as I can remember, the installation was a plain Raspbian one, 
that I then installed Octoprint on i.e. not OctoPi.

The octoprint.log is almost empty:

~/.octoprint/logs $ cat octoprint.log
2017-01-22 09:43:25,882 - octoprint.server.util - INFO - New connection 
from client: 82.28.50.184

There are many other logs with a date on the end of the file name,  I've 
had a look at a few of these, but they just contain client connection 
info, no version number etc.

Is there a string I could grep for in them that would help locate the 
version number?

-rw-r--r-- 1 pi pi   720 Dec 30 18:49 octoprint.log.2016-12-29
-rw-r--r-- 1 pi pi   997 Dec 31 19:40 octoprint.log.2016-12-30
-rw-r--r-- 1 pi pi   802 Jan  1 13:29 octoprint.log.2016-12-31
-rw-r--r-- 1 pi pi  1259 Jan  2 20:29 octoprint.log.2017-01-01
-rw-r--r-- 1 pi pi  1436 Jan  3 22:55 octoprint.log.2017-01-02
-rw-r--r-- 1 pi pi   540 Jan  4 19:41 octoprint.log.2017-01-04
-rw-r--r-- 1 pi pi   998 Jan  6 08:31 octoprint.log.2017-01-05
-rw-r--r-- 1 pi pi  1800 Jan  7 08:56 octoprint.log.2017-01-06
-rw-r--r-- 1 pi pi   622 Jan  7 22:16 octoprint.log.2017-01-07
-rw-r--r-- 1 pi pi   900 Jan  9 09:28 octoprint.log.2017-01-08
-rw-r--r-- 1 pi pi  1717 Jan 10 13:33 octoprint.log.2017-01-09
-rw-r--r-- 1 pi pi  1078 Jan 11 13:46 octoprint.log.2017-01-10
-rw-r--r-- 1 pi pi   802 Jan 12 14:57 octoprint.log.2017-01-11
-rw-r--r-- 1 pi pi  1260 Jan 13 14:07 octoprint.log.2017-01-12
-rw-r--r-- 1 pi pi  1355 Jan 14 19:04 octoprint.log.2017-01-13
-rw-r--r-- 1 pi pi   540 Jan 15 19:58 octoprint.log.2017-01-14
-rw-r--r-- 1 pi pi   540 Jan 16 20:16 octoprint.log.2017-01-15
-rw-r--r-- 1 pi pi   622 Jan 17 19:38 octoprint.log.2017-01-16
-rw-r--r-- 1 pi pi   720 Jan 18 19:18 octoprint.log.2017-01-17
-rw-r--r-- 1 pi pi  1260 Jan 20 00:54 octoprint.log.2017-01-19
-rw-r--r-- 1 pi pi   719 Jan 20 22:43 octoprint.log.2017-01-20
-rw-r--r-- 1 pi pi  1259 Jan 22 00:29 octoprint.log.2017-01-21
-rw-r--r-- 1 pi pi     0 Sep  9  2014 serial.log
(oprint)pi@3DPrinterRaspi ~/.octoprint/logs $

If it's not possible to resolve this, I am happy to do a re-install, but 
would rather keep Raspian (Wheezy) and just load in the latest 
OctoPrint. If I back up ~/.octoprint is it simply a case of doing a 
fresh install and then copying back all the files back into 
~/.octoprint? I assume I would just follow the instructions on the Wiki 
at 
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian 
?

Regards,

Jeremy

foosel, 2017-01-24:
I'm an idiot. Shouldn't have been

```
setuptools --version
```

but 

```
easy_install --version
```

Sorry for that.

Regarding a string to grep for, now that I think about it a bit more, if you still were on OctoPrint 1.1.x before the upgrade I'm not even sure that already logged its version on startup. As I said, a lot of water has gone down the river since that version was current.

I think that a backup + fresh install will probably really be the best approach though. Chances are high that this problem is unsolvable anyhow, due to "distribute", the ancient Python version and even if we could now get it to run, chances are high that a future update of any of the third party dependencies OctoPrint relies on, or OctoPrint itself, would break things again.

I'd strongly suggest to not go with Raspbian Wheezy if you want to build it yourself but at least use the Jessie Lite image though - otherwise you have the next outdated environment waiting to cause problems. Following the setup guide should get you going. Don't know though why you want to do that to yourself instead of just flashing OctoPi, way faster ;)

But yes, just securing everything under ~/.octoprint and then copying that over should work usually. OctoPrint has always stored all its settings and data files in that folder. Make sure maybe that you didn't do any weird stuff with regards to symlinks or whatnot ;)

foosel, 2017-01-24:
As a side note, you *might* also want to try an upgrade of your OS (as in, `apt-get update`, `apt-get upgrade` or even `apt-get dist-upgrade`), that could also get you a fresher environment, but I have NO idea what that might do, so if you attempt to do that, better do the backup of ~/.octoprint first.

m0rgano, 2017-01-24:
Hi Gina,

(oprint)pi@3DPrinterRaspi ~/.octoprint/logs $ easy_install --version
distribute 0.6.24

It sounds like the easiest route will be to flash the latest Octopi 
image. I only use this particular Raspberry Pi with my UM2 and a 
Microsoft USB Webcam, so hopefully all of that will work out of the box now.

Thanks for your help.

Jeremy

m0rgano, 2017-01-24:
Hi Gina,

Yes, I ran the OS upgrade first before I attempted to upgrade OctoPrint. 
Maybe it would have been better the other way round...? :)

Regards,

Jeremy

foosel, 2017-01-24:
Whether you run the upgrade before or after shouldn't matter, but I guess that just wasn't sufficient then either.

Any objections to closing this now that you've decided on the backup-and-reinstall-approach?

m0rgano, 2017-01-24:
Hi Gina,

Yes, please close the issue.

Jeremy

closed 2017-01-24 by foosel

ldm314, 2017-02-06:
Commenting on this as it seems to be the only thread relevant from a search. I have ran into the same issue trying to do a clean install. I started on a Pi B 2.0 that I had a 3 year old install on. Ran apt-get update and apt-get dist-upgrade.

Following the install doc I have ran into the same error:
<pre>
Installed /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg
Processing dependencies for OctoPrint==1.3.1
Searching for webassets>=0.10
Reading http://pypi.python.org/simple/webassets/
Best match: webassets 0.12.1
Downloading https://pypi.python.org/packages/9a/35/d86c6d3bd31e2e9c8351c8766ef3688c75d56bc5233ed49061e522efc3cc/webassets-0.12.1.tar.gz#md5=429401824556e6e0868e437f9e561ba6
Processing webassets-0.12.1.tar.gz
Running webassets-0.12.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-zbPsFX/webassets-0.12.1/egg-dist-tmp-K_z6VR
error: /tmp/easy_install-zbPsFX/webassets-0.12.1/src/webassets/__init__.py: No such file or directory
</pre>

ldm314, 2017-02-06:
Also, installing webassets manually via ./venv/bin/pip install webassets gets past this issue.

Patman27, 2017-02-20:
Thank you @ldm314 ! I was caught up on this issue as well. After installing webassets manually and re-installing OctoPrint, it worked for me too.

Burdalfis, 2017-03-02:
./venv/bin/pip install webassets

This is fantastic, Fixed my issues installing on an Intel Edison running ubilinux, which development for has been dropped so a while ago so it's outdated.

sekarpdkt, 2017-07-24:
./venv/bin/pip install webassets

This is fantastic, Fixed my issues during fresh install on Radxa rock running radxa_rock_pro_debian_wheezy_150128_nand.img

Note: Not sure why, but when i installed it from ./venv/bin/ directory, it was giving issue. I did it from OctoPrint directory and it works. Might help others

cscott, 2017-07-26:
`~/oprint/bin/pip install webassets` worked for me to get around this problem.

ThomasSc, 2017-07-29:
for me it works with
./venv/bin/pip install --upgrade webassets

castaway, 2018-02-26:
This works for me too, tho it complains a lot about SSL/TSL not being secure..