OctoPrint/OctoPrint #1728: [Request] Keep temperature when cancelling print job

henrikkoch, 2017-01-22:
I would like a feature that enables the printer to keep the current settings of bed and nozzle temp. even though you cancel a print job. In that way you can start another job right after if you by any reason had to stop the current job. In the current setup depending of how fast you are on your fingers in between two jobs, then you have to way for mostly the bed going up in temp again. Thats a little "crazy" just to wait several minutes if you just want to continue with a new job.

AlessioMorale, 2017-01-22:
You can change that behaviour by editing the following setting:
Settings>GCODE Scripts> After Print Job is Cancelled
specifically you need to remove one or both the following lines:
`{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}`

obviously you may want to consider the safety aspect of this change

henrikkoch, 2017-01-23:
Hi and thanks for your info. Due to the safety this is way I ned a button

Med venlig hilsen
Henrik J. Koch

foosel, 2017-01-23:
As pointed out, you can already modify [the cancel script](http://docs.octoprint.org/en/master/features/gcode_scripts.html). And you can also define [custom controls](http://docs.octoprint.org/en/master/features/custom_controls.html) to have your own safety all-heater-shutoff button. So this is already possible just fine with what's available in OctoPrint.

If you want to get fancy you could also write a plugin that implements some timer functionality ("shut off stuff if no user input for 5min after print got cancelled").

closed 2017-01-23 by foosel

thewavelength, 2019-06-04:
@AlessioMorale thanks for pointing this out. What would be the alternative if I actually want to keep the temperatures after a print finished successfully? Sadly, the GCODE that is being inserted for cancelled prints is not there for successful prints thus I am not able to remove it. I also had a look at my GCODE-file itself and couldn't find any command to specifically shutdown the extruder and bed. Is this part of the firmware and because of that not avoidable? Thank you!