OctoPrint/OctoPrint #1729: Exception Error at end of print

GingerNutt, 2017-01-23:
#### What were you doing?

A print completed.

#### What did you expect to happen?

Nothing

#### What happened instead?

Exception Error - every time

#### Branch & Commit or Version of OctoPrint

1.3.0 Master Branch

#### Printer model & used firmware incl. version

Wanhao i3 v2.1

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 55.0.2883.87 m

#### Link to octoprint.log

https://gist.github.com/GingerNutt/f15551737023288584bb30a541440d2e

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

[image]

GingerNutt, 2017-01-23:
Updated.

nophead, 2017-01-23:
Your gcode has an error in it:

N21 G1 F{travel_speed}*91

{travel_speed} should have been replaced by a number by the Slicer but it
must not match what the Slicer looks for.

foosel, 2017-01-23:
What @nophead said. That doesn't appear to be the actual issue for your exception though, but I can't say for sure since the terminal log you provided was not from when the problem occurred. Please reproduce the problem and then provide the terminal output from the actual issue.

In any case, it looks like something at the end of your print makes your printer stop replying, which is why OctoPrint runs into a timeout. Its attempts to get the communication going again fail repeatedly, so all OctoPrint can do is give up and make you aware of this problem.

But again, to be able to tell you what specifically might be causing this, a communication log (serial.log or terminal output at time of occurrence) is mandatory.

GingerNutt, 2017-01-23:
https://gist.github.com/GingerNutt/70c5274d627bc45a2aa7bd7836267d38

foosel, 2017-01-23:
```
--- too many lines to buffer, cut off ---
```

You had the "Autoscroll" option set to off sometime before the end of the print, right? That sadly made that log incomplete too since the allowed buffer of log lines in the terminal window ran full before the actual print end was reached from the looks of it.

Please disable that option, reproduce the issue, provide the untruncated end of the print log.

If you can't monitor the terminal for this exact point in time due to whatever reason, the easiest way to achieve this might actually be by first enabling "Settings > Serial > `serial.log`", doing a small print and then providing the `serial.log` (as available under "Settings > Logs") once the print finishes.

In a nutshell, what's needed here are the last 20 or so lines leading up to the first "communication timeout" message, and everything from then on until the final "Too many consecutive timeouts".

GingerNutt, 2017-01-23:
https://gist.github.com/GingerNutt/9ef6fca4d37e308a8e1df0f34dad9ea6

Where do I find the serial log? I switched it on.

foosel, 2017-01-23:
```
Send: N16815 G4 P120000*85
```

This is your problem. You are telling your printer to effectively go dead there for 2min straight. It stops responding during that time completely and cannot be controlled via serial at all. It's effectively dead to OctoPrint (or any host). The problem with this particular printer and firmware is that contrary to how that command actually should be handled (with the printer initiating the dwell when receiving it, then replying with `ok` once the time is over) the firmware of your printer instead handles that command differently to a point that it's impossible for OctoPrint to determine that the printer is processing a dwell command now.

You have two options, you can remove that line from your end GCODE (I don't know if that comes from the one in your slicer or in OctoPrint itself), or you can increase OctoPrint's timeout setting to be higher than 2min (which means that if something in your printer communication goes really wrong, the chances are higher that your print gets a blob while OctoPrint waits for the timeout to pass before attempting to tickle the printer back into action).

In any case, this makes it a duplicate of #1506, so I'm closing this now and would ask you to rather follow that other ticket.

closed 2017-01-23 by foosel