OctoPrint/OctoPrint #1730: Having same problem as m0rgano commented Jan 22, 2017, only on original install not on upgrade.

llabarrec, 2017-01-24:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box). Do not delete any
lines from the template but those enclosed in [ and ] - and those please
DO delete, they are only provided for your information and removing them
makes your ticket more readable :)

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

llabarrec, 2017-01-24:
I am having the same problem as m0rgano commented Jan 22, 2017. only on original install not on upgrade.  
./venv/bin/python setup.py install
...
Searching for webassets>=0.10
Reading http://pypi.python.org/simple/webassets/
Best match: webassets 0.12.1
Downloading https://pypi.python.org/packages/9a/35/d86c6d3bd31e2e9c8351c8766ef3688c75d56bc5233ed49061e522efc3cc/webassets-0.12.1.tar.gz#md5=429401824556e6e0868e437f9e561ba6
Processing webassets-0.12.1.tar.gz
Running webassets-0.12.1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-dttzFI/webassets-0.12.1/egg-dist-tmp-on2TTU
error: /tmp/easy_install-dttzFI/webassets-0.12.1/src/webassets/__init__.py: No such file or directory

foosel, 2017-01-24:
Then please do NOT open a new ticket but comment on the existing one...

closed 2017-01-24 by foosel