OctoPrint/OctoPrint #1731: [Request] Token based authentication

MiggyMan, 2017-01-24:
Would it be possible (or feasible) to add token based authentication system to the octoprint ui, either automatically generated per user or salted based on a provided passphrase and username (and something nice and painfully long) ?

For example

http://octopussy:5000?token=WW91IGhhdmUgc2V4IHdpdGggY2FrZQo=

To log me in as the user with this token.

The main "need" for this is probably not going to be that common, at the moment i'm running octoprint on a p4 machine with a 15" touch screen and while persistent sessions are workable having to login after restarting the machine for updates is a little "painful" :)

foosel, 2017-01-24:
For your specific use case, maybe `autologinLocal` and friends might already solve your problem? See the [docs](http://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control).

MiggyMan, 2017-01-24:
That might actually work, ta, i'll have a dig when I get home :)

MiggyMan, 2017-01-24:
Though as a side note I wonder it there might be a use for token based auth for people giving out access to others ?

foosel, 2017-01-24:
Which is why I haven't closed this, merely pointed you in a direction that might already help you :) Let's see if there's any more interest by the general community first though. 

In theory it might be interesting to implement something similar to the "authenticated application" feature common in a lot of web applications, but then again the question is if there's really demand for that since it makes the whole authentication stuff even more complex. And we already do have local auto-login, per-user api keys and also [application session keys](http://docs.octoprint.org/en/master/api/apps.html#apps).

MiggyMan, 2017-01-24:
It would allow people to implement alternate authentication systems without having to bug you about it :D

foosel, 2017-01-24:
I'd rather offer a plugin interface for that ;)

MiggyMan, 2017-02-26:
Sorry for not getting back sooner, life/work (mostly work wrecking life tbh) got in the way.

Autologin works like a charm :)

LazeMSS, 2021-03-31:
Closing - fixed by using Autologin

closed 2021-03-31 by LazeMSS

foosel, 2021-03-31:
And `accessControl.{trustRemoteUser|remoteUserHeader|addRemoteUsers}`