OctoPrint/OctoPrint #1732: Mismatch in printer profile API documentation and behaviour

Klipi, 2017-01-25:
#### What were you doing?

I'm trying to use the printer profile API, but the current [documentation](http://docs.octoprint.org/en/1.3.1/api/printerprofiles.html?highlight=printerprofile#get--api-printerprofiles) seems to be inaccurate. I'd love to use the API as documented, if possible, for easier parsing.

The exact command used was 
`curl -H "X-Api-Key: key" http://octopi.local/api/printerprofiles`

#### What did you expect to happen?

The documentation has an example response with an array of Profile-type objects:
```
HTTP/1.1 200 OK
Content-Type: application/json

[
  {
    "id": "_default",
    "name": "Default",
    "color": "default",
    "model": "Generic RepRap Printer",
    "default": true,
    "current": true,
    "resource": "http://example.com/api/printerprofiles/_default",
    "volume": {
      "formFactor": "rectangular",
      "origin": "lowerleft",
      "width": 200,
      "depth": 200,
      "height": 200
    },
    "heatedBed": true,
    "axes": {
      "x": {
        "speed": 6000,
        "inverted": false
      },
      "y": {
        "speed": 6000,
        "inverted": false
      },
      "z": {
        "speed": 200,
        "inverted": false
      },
      "e": {
        "speed": 300,
        "inverted": false
      }
    },
    "extruder": {
      "count": 1,
      "offsets": [
        {"x": 0.0, "y": 0.0}
      ]
    }
  },
  {
    "id": "my_profile",
    "name": "My Profile",
    "color": "default",
    "model": "My Custom Printer",
    "default": false,
    "current": false,
    "resource": "http://example.com/api/printerprofiles/my_profile",
    "volume": {
      "formFactor": "rectangular",
      "origin": "lowerleft",
      "width": 200,
      "depth": 200,
      "height": 200
    },
    "heatedBed": true,
    "axes": {
      "x": {
        "speed": 6000,
        "inverted": false
      },
      "y": {
        "speed": 6000,
        "inverted": false
      },
      "z": {
        "speed": 200,
        "inverted": false
      },
      "e": {
        "speed": 300,
        "inverted": false
      }
    },
    "extruder": {
      "count": 1,
      "offsets": [
        {"x": 0.0, "y": 0.0}
      ]
    }
  },
]
```

#### What happened instead?

I get an object with the profiles as attributes: profiles.<profile id>
```
{
  "profiles": {
    "_default": {
      "axes": {
        "e": {
          "inverted": false, 
          "speed": 300
        }, 
        "x": {
          "inverted": false, 
          "speed": 6000
        }, 
        "y": {
          "inverted": false, 
          "speed": 6000
        }, 
        "z": {
          "inverted": false, 
          "speed": 200
        }
      }, 
      "color": "default", 
      "current": false, 
      "default": false, 
      "extruder": {
        "count": 1, 
        "nozzleDiameter": 0.4, 
        "offsets": [
          [
            0.0, 
            0.0
          ]
        ]
      }, 
      "heatedBed": true, 
      "id": "_default", 
      "model": "Generic RepRap Printer", 
      "name": "Default", 
      "resource": "http://octopi.local/api/printerprofiles/_default", 
      "volume": {
        "custom_box": false, 
        "depth": 200.0, 
        "formFactor": "rectangular", 
        "height": 200.0, 
        "origin": "lowerleft", 
        "width": 200.0
      }
    }, 
    "prenta": {
      "axes": {
        "e": {
          "inverted": false, 
          "speed": 300
        }, 
        "x": {
          "inverted": false, 
          "speed": 6000
        }, 
        "y": {
          "inverted": false, 
          "speed": 6000
        }, 
        "z": {
          "inverted": false, 
          "speed": 200
        }
      }, 
      "color": "default", 
      "current": true, 
      "default": true, 
      "extruder": {
        "count": 1, 
        "nozzleDiameter": 0.4, 
        "offsets": [
          [
            0.0, 
            0.0
          ]
        ]
      }, 
      "heatedBed": true, 
      "id": "prenta", 
      "model": "", 
      "name": "Prenta", 
      "resource": "http://octopi.local/api/printerprofiles/prenta", 
      "volume": {
        "custom_box": false, 
        "depth": 200.0, 
        "formFactor": "rectangular", 
        "height": 200.0, 
        "origin": "lowerleft", 
        "width": 200.0
      }
    }, 
    "um2g": {
      "axes": {
        "e": {
          "inverted": false, 
          "speed": 300
        }, 
        "x": {
          "inverted": false, 
          "speed": 6000
        }, 
        "y": {
          "inverted": false, 
          "speed": 6000
        }, 
        "z": {
          "inverted": false, 
          "speed": 200
        }
      }, 
      "color": "default", 
      "current": false, 
      "default": false, 
      "extruder": {
        "count": 1, 
        "nozzleDiameter": 0.4, 
        "offsets": [
          [
            0.0, 
            0.0
          ]
        ]
      }, 
      "heatedBed": false, 
      "id": "um2g", 
      "model": "", 
      "name": "UM2G", 
      "resource": "http://octopi.local/api/printerprofiles/um2g", 
      "volume": {
        "custom_box": false, 
        "depth": 120.0, 
        "formFactor": "rectangular", 
        "height": 115.0, 
        "origin": "lowerleft", 
        "width": 120.0
      }
    }
  }
}
```

#### Branch & Commit or Version of OctoPrint

Version: 1.3.1 (rc/maintenance branch) 

#### Browser and Version of Browser, Operating System running Browser

macOS Sierra 10.12.2 (16C67), curl 7.51.0 (x86_64-apple-darwin16.0) libcurl/7.51.0 SecureTransport zlib/1.2.8

#### Link to octoprint.log

http://pastebin.com/dMs7yTDg

I have read the FAQ.

foosel, 2017-01-26:
Thanks for the heads-up. Sadly we'll have to go with what's implemented here already and adjust the docs accordingly instead of the other way around, since there are a couple of clients out there which depend on the API staying backwards compatible.

Will fix the docs.

foosel, 2017-03-06:
Solved via the above commit, fixed on `maintenance` and soon `devel` branch, will be part of the 1.3.2 release.

commit 01cffafe (2017-03-06): Fixed documentation of printer profile API

Closes #1732
closed 2017-03-16 by foosel