OctoPrint/OctoPrint #1733: After upgrade to 1.3.1 no automatic start

SCiunczyk, 2017-01-25:
#### What were you doing?

I made upgrade to 1.3.1 from OctoPrint UI.  As requested system was rebooted.

#### What did you expect to happen?

Octoprint should start automatically as usual.

#### What happened instead?

I received error: 
503 Service Unavailable
No server is available to handle this request. 

#### Branch & Commit or Version of OctoPrint

Version: 1.3.1 (master branch)

#### Printer model & used firmware incl. version

Probably not applicable ... but RepRap Prusa i3 with Marlin 1.1.0-RC7

#### Browser and Version of Browser, Operating System running Browser

Win 10, Firefox

#### Link to octoprint.log

[octoprint.log](https://gist.github.com/SCiunczyk/4de02242fa7e6d8dfce695c261f3b3fb)

#### Link to contents of terminal tab or serial.log

not a communication problem

#### Link to contents of Javascript console in the browser

not applicable

#### Screenshot(s) showing the problem:

not applicable

#### Environment info

OctoPrint is installed on Nano Pi NEO board with latest Armbian 5.20 Jessie 3.4.112. Was working perfectly for last two months.

Now I can start it manually with command: *~/OctoPrint/venv/bin/octoprint serve*
but automatic start or via: *sudo service octoprint start* doesn't work.

I have read the FAQ.

foosel, 2017-01-26:
Sounds like it would be caused by what's already tracked in #1657. You need to fix the init script under /etc/init.d/octoprint, the one shipped with earlier versions of OctoPrint contained a bug that causes this kind of issue. I cannot do that update automatically for you via the usual update process. However, I did adjust the release notes for 1.3.1 and the release announcement on the OctoBlog to give people a stronger heads-up regarding this.

SCiunczyk, 2017-01-26:
Hi Gina, you were right. After applying octoprint.ini from  [42e3922](https://github.com/foosel/OctoPrint/commit/42e39220534ceb47890564f7b9305340acc47776) autostart is working again.

Many thanks. 

BTW I would like to recommend NanoPi NEO as very good board for OctoPrint - only 40x40mm in size :-)

foosel, 2017-01-26:
I hope you mean octoprint.init from e121569 :) Because 42e3922 was a code change :D

NanoPi NEO sounds cute!

SCiunczyk, 2017-01-26:
Yes, you right. I don't know why I put this link...

foosel, 2017-01-26:
It's mentioned in the same ticket. And no harm done, just wanted to make sure.

Closing this now if you don't mind. I hope that the big heads-up in the release letter and the release announcement post I put there this morning will help other people that stumble over this.

Still bothers me that I can't take care of that automatically, but I really don't want to start messing with people's systems on that level.

closed 2017-01-26 by foosel

SCiunczyk, 2017-01-26:
Not a problem at all. If we can manage this manually it's fine. I think that all makers should have a little knowledge of system tuning, and situation like this keep us vigilant ;-)

Have a nice evening Gina.
Slawek