OctoPrint/OctoPrint #1734: Unable to delete printer profiles via the trashcan icon.

ayourk, 2017-01-25:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box). Do not delete any
lines from the template but those enclosed in [ and ] - and those please
DO delete, they are only provided for your information and removing them
makes your ticket more readable :)

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

Awhile back I added a printer I intended on purchasing.  I never really did purchase this printer, so I wanted to delete that printer profile from OctoPrint.  OctoPrint does not register (as far as I can tell) that the printer profile is deleted.

To recreate:  Create a printer profile.  Try to delete that profile.  Deletion fails.

#### What did you expect to happen?
I expected the list of printer profiles to refresh without the printer profile I chose to delete.
#### What happened instead?
OctoPrint appeared to process something, then nothing in the list changed.
#### Branch & Commit or Version of OctoPrint
OctoPrint v1.3.1, but this has been happening for several versions now.
OctoPrint is running on Ubuntu Mate 16.04 on a Raspberry Pi 3.
#### Printer model & used firmware incl. version
Original Prusa i3 MK2 with 1.3.10a2 firmware (customized)

#### Browser and Version of Browser, Operating System running Browser
I am using Google Chrome (stable) on Ubuntu Linux 16.04 LTS.

#### Link to octoprint.log
https://gist.github.com/ayourk/ae91f9af56b194ef1d42aabe7a4c988f

#### Link to contents of terminal tab or serial.log
Not relevant.

#### Link to contents of Javascript console in the browser
https://gist.github.com/ayourk/361b1333c8305854ee7b8a8404308f21

I have read the FAQ.

foosel, 2017-02-15:
If it is the printer profile with the identifier "_default" you are trying to delete, that is not a bug but a misguided feature that I'll have to change. 

Sadly your log does not appear to contain an actual deletion attempt (or I'm just blind right now) so I can't verify that right now - could you please elaborate?

ayourk, 2017-02-15:
Here is a picture of my Printer Profiles:  http://i.imgur.com/aK4VxMp.png

I'm trying to delete the MakerGear M2 from the list.  The interface appears to go through the motions of attempting to delete the printer profile, but the listing never disappears.

foosel, 2017-02-17:
Sadly that screenshot doesn't help to determine if the profile in question has the `_default` identifier mentioned above. Could you please check what it says when you click on that MakerGear M2 profile's little pencil icon to edit it in the Identifier field?

[image]

If it says `_default` like in the above screenshot, it's as I thought and you are trying to delete what you defined as the default profile, and that currently can't be deleted. Has to be changed, was a stupid design decision when I introduced printer profiles.

ayourk, 2017-02-17:
I can confirm it says "_default" in the Identifier field.  How do I change that?

ayourk, 2017-02-17:
I would have thought that the "STAR" symbol followed the default printer profile.

foosel, 2017-02-17:
You currently can't change that. Identifiers are not modifiable. As I said, I made a wrong design decision here. The "star" does follow the default printer profile, the problem rather is that the profile with the identifier "_default" (both are NOT necessarily the same) is stored differently than the others and currently doesn't support deletion. It's also not something I can easily fix because that would cause problems in other places. So that might take a while to allow, sorry.

ayourk, 2017-02-17:
I think I fixed it on my end by manually editing config.yaml while octoprint was shut down.

closed 2017-02-17 by ayourk

necro-nemesis, 2019-12-15:
Problem appears to remain with deleting the "_default" printer.

Michri, 2020-02-10:
I still have this issue with Octoprint 1.3.12. I cannot delete any printer, even these ones not being _default:

[image]

[image]

[image]

LOG:

2020-02-10 07:38:04,465 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 284, in on_api_get
    if refresh_notices or not self._is_notices_cache_valid():
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 865, in _is_notices_cache_valid
    return mtime + self._notices_cache_ttl >= time.time() > mtime
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
2020-02-10 07:38:04,471 - tornado.access - ERROR - 500 GET /api/plugin/pluginmanager (::ffff:192.168.178.72) 12.14ms
2020-02-10 07:38:04,652 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 0.16s
2020-02-10 07:38:04,997 - octoprint.plugins.announcements - INFO - Loaded channel _releases from https://octoprint.org/feeds/releases.xml in 0.14s
2020-02-10 07:38:05,815 - octoprint.plugins.announcements - INFO - Loaded channel _blog from https://octoprint.org/feeds/octoblog.xml in 0.14s
2020-02-10 07:38:06,018 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from https://plugins.octoprint.org/feed.xml in 0.14s
2020-02-10 07:38:06,250 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 0.14s
2020-02-10 07:38:08,754 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.178.72
2020-02-10 07:38:14,789 - tornado.access - WARNING - 404 DELETE http://192.168.178.25/api/printerprofiles/anycubic_4max_pro (::ffff:192.168.178.72) 9.32ms
2020-02-10 07:42:33,060 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.178.72
2020-02-10 07:42:36,433 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.178.72
2020-02-10 07:42:37,433 - octoprint.server.util.flask - INFO - Passively logging in user dummy from ::ffff:192.168.178.72
2020-02-10 07:42:38,739 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 284, in on_api_get
    if refresh_notices or not self._is_notices_cache_valid():
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 865, in _is_notices_cache_valid
    return mtime + self._notices_cache_ttl >= time.time() > mtime
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
2020-02-10 07:42:38,746 - tornado.access - ERROR - 500 GET /api/plugin/pluginmanager (::ffff:192.168.178.72) 13.10ms
2020-02-10 07:44:03,153 - octoprint.plugins.errortracking - INFO - Initialized error tracking
2020-02-10 07:44:47,550 - tornado.access - WARNING - 404 DELETE http://192.168.178.25/api/printerprofiles/anycubic_4max_pro (::ffff:192.168.178.72) 9.22ms
2020-02-10 07:52:29,197 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-02-10 08:04:21,707 - tornado.access - WARNING - 404 DELETE http://192.168.178.25/api/printerprofiles/dhdghh (::ffff:192.168.178.72) 11.63ms
2020-02-10 08:07:29,199 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-02-10 08:15:27,101 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.178.72
2020-02-10 08:15:29,709 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.178.72
2020-02-10 08:15:29,792 - octoprint.server.util.flask - INFO - Passively logging in user dummy from ::ffff:192.168.178.72
2020-02-10 08:15:30,805 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 284, in on_api_get
    if refresh_notices or not self._is_notices_cache_valid():
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 865, in _is_notices_cache_valid
    return mtime + self._notices_cache_ttl >= time.time() > mtime
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
2020-02-10 08:15:30,841 - tornado.access - ERROR - 500 GET /api/plugin/pluginmanager (::ffff:192.168.178.72) 41.93ms
2020-02-10 08:22:29,200 - octoprint.server.heartbeat - INFO - Server heartbeat <3

I am surprised about the huge amount of errors.

Any idea?