OctoPrint/OctoPrint #1735: [Brainstorming] Some suggestions for language in source files

seigel, 2017-01-27:
#### What were you doing?

Reading the instructions in a couple of files.  Nothing serious, just some tweaks.

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint
master

#### Printer model & used firmware incl. version
N/A
#### Browser and Version of Browser, Operating System running Browser
N/A
#### Link to octoprint.log
N/A
#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s) showing the problem:
N/A
I have read the FAQ.
YES

[changes.txt](https://github.com/foosel/OctoPrint/files/734525/changes.txt)

foosel, 2017-02-17:
Sorry, that somehow got buried. I like those changes! Why didn't you create a PR instead of a Brainstorming ticket? And do you maybe still want to do that? :)

foosel, 2017-03-06:
Applied them in the above commit. Available on `maintenance`, soon `devel`, will be part of 1.3.2.

seigel, 2017-03-06:
Thanks sorry, I missed your response.

foosel, 2017-03-16:
1.3.2 was just released.

closed 2017-03-16 by foosel