OctoPrint/OctoPrint #1736: Security Issue - Print to any printer without credentials

h3ku, 2017-01-27:
#### What were you doing?

Searching for some security issues.

#### What did you expect to happen?

Find some critical security issues.

#### Branch & Commit or Version of OctoPrint

Master - Up to Date

#### Screenshot(s) showing the problem:

I found this "bug" inspecting the WebSockets communication between the printer and de frontend, basically, I find that in the first WebSocket incoming request the server sends the API key in clear text... 
[image]

As you can see with this API key a malicious user can do whatever he wants with the printer, like print some massive dildos for example.
[image]

h3ku, 2017-01-27:
The extra information it's not necessary to reproduce the bug.

foosel, 2017-01-27:
It's an anonymous API key only that will require an existing login session to do anything with that goes beyond anonymous access.

foosel, 2017-01-27:
To demonstrate what I meant above:

[image]

```
$ curl -i -H "X-Api-Key: D11EBBA29CC946F5A1223E82DA216DBB" -H "Accept: application/json" -H "Content-Type: application/json" -X POST -d '{"command": "pause"}' http://localhost:5 000/api/job
HTTP/1.1 401 UNAUTHORIZED
Content-Type: text/html
Content-Length: 448
Cache-Control: pre-check=0, no-cache, no-store, must-revalidate, post-check=0, max-age=0
Pragma: no-cache
Expires: -1
X-Clacks-Overhead: GNU Terry Pratchett
Set-Cookie: session_P5000=eyJfaWQiOnsiIGIiOiJOalJsTm1GaFpqSTVNbU5qWVdJM05qRTBZVFF3TlRjMllXWTFNVFJqTkRjPSJ9fQ.C2zd6g.2RGDVZAlgYYdzaRh6s48LXuaoCc; Path=/; HttpOnly
Server: TornadoServer/4.0.2

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>401 Unauthorized</title>
<h1>Unauthorized</h1>
<p>The server could not verify that you are authorized to access the URL requested.  You either supplied the wrong credentials (e.g. a bad password), or your browser doesn't understand how to supply the credentials required.</p><p>In case you are allowed to request the document, please check your user-id and password and try again.</p>
```

Compare that with an actual API key as configured for this particular instance:

```
$ curl -i -H "X-Api-Key: test1234" -H "Accept: application/json" -H "Content-Type: application/json" -X POST -d '{"command": "pause"}' http://localhost:5000/api/job
HTTP/1.1 204 NO CONTENT
Content-Type: text/html; charset=utf-8
Content-Length: 0
Cache-Control: pre-check=0, no-cache, no-store, must-revalidate, post-check=0, max-age=0
Pragma: no-cache
Expires: -1
X-Clacks-Overhead: GNU Terry Pratchett
Set-Cookie: session_P5000=.eJyrVopPK0otzlCyKikqTdVRis9MUbKqVlJIUrJS8ssKyvHLdcuIyvI09c31y4oM9zT2y3I1iAwJLPcLSTaKDI809A0JyvJzSbZVqtVRykxJzSvJLKnUSywtyYgvqSxIVbLKK83JQZJBMj3CyK08MdAWrLO0OLUoHqtcLQCmITPz.C2zd1g.hroAMhYStYoTUcYVu8MFRPMNG8I; Path=/; HttpOnly
Server: TornadoServer/4.0.2
```

That so called UI API key you are seeing there is actually something that currently is being discussed to be removed since in fact it only has cosmetic value and doesn't authenticate anything. 

So this is not in fact a security issue, even though on first glance without an attempt to actually exploit it it might look like one, but in fact an implementation quirk showing an "API key" with which you can do pretty much nothing. It merely allows you to access the API with the same level of access as an anonymous (= not logged in) visitor.

Sorry to disappoint.

closed 2017-01-27 by foosel