OctoPrint/OctoPrint #1738: Cannot delete gcode files with 1.3.2.dev24+g54a9a7b (maintenance branch)

KevMags, 2017-01-28:
#### What were you doing?

Trying to delete  uploaded files by using the trash can button to the right of the filename.  Both local and SD card files.
#### What did you expect to happen?

The file to delete.

#### What happened instead?

Circle animation on trash icon but file does not delete.  Have to refresh page to stop the animation in icon.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.2.dev24+g54a9a7b (maintenance branch)

#### Printer model & used firmware incl. version

Maker Select i3 and 2 Prusa i3s,  Marlin RC8 

#### Browser and Version of Browser, Operating System running Browser

Safari 10.1, macOS 10.12.4

#### Link to octoprint.log

https://gist.github.com/KevMags/a28027aaa44ea18615f023e06f457129

#### Link to contents of terminal tab or serial.log

https://gist.github.com/KevMags/4a0fd08a781adadd3d7cc918b1ae7eb7

#### Link to contents of Javascript console in the browser

https://gist.github.com/anonymous/b8f2115e50e3658d73a70a3577b985c4

#### Screenshot(s) showing the problem:

I have read the FAQ.

foosel, 2017-02-15:
That should be fixed on `maintenance` and will also be merged to `devel` ASAP, thanks for reporting! 👍

KevMags, 2017-02-15:
Thanks for all of your effort to make a great piece of software.

foosel, 2017-03-16:
1.3.2 was just released.

closed 2017-03-16 by foosel