OctoPrint/OctoPrint #1740: Prints Freezing and Leaving Heaters on in 1.3.1

hebs, 2017-01-31:
I am in a large 3d printing channel on slack, specifically for drone racing. Several of us use octoprint, and have been for quite a while. Everyone updated to 1.3.1 and I have at least 5 people reporting prints freezing and leaving heaters on. I didnt think much of it until it happened to me, twice. I have put about 30kg through this machine, with octoprint, same fw, same hardware, etc... and this is the first time ive experienced anything like this. 

Terminal output: https://github.com/guysoft/OctoPi/issues/303

hebs, 2017-01-31:
I am running an azteeg x5mini 1.3 with smoothieware, but the other guys having this issue are using various firmware and electronics.

foosel, 2017-01-31:
The error in your terminal output is caused a level below OctoPrint, most commonly by cabling or power issues. There's nothing I can do about this kind of error, it's basically the serial port breaking away below OctoPrint's feet. See also the long discussion and further explanation of this in eg #1603.

In order to be able to do any kind of analysis on this however and to rule out anything in OctoPrint being the culprit (or to verify it is, because so far it looks like what I said above) a **fully filled out ticket template** in this ticket here is required by everyone affected by it. That means logs, *especially* the serial.log, information regarding hardware and firmware (printer electronics, hardware OctoPrint is running on), and also information what of the steps mentioned in the above linked ticket have already been attempted, how often the issue manifests, with which exact version it started manifesting (it might have started with a version you skipped for example) and so on. Basically: the more information the better.

I can't help you when running blind, especially not considering that this is the first report I've gotten about any such problems with 1.3.1. As a side note, there were no changes in OctoPrint's communication layer between 1.3.0 and 1.3.1 at all, reducing the likelihood of this being a problem introduced in 1.3.1.

PS: currently out sick, so once you've provided the requested information it might still take a bit for me to look into it.

opuscroakus, 2017-02-01:
I haven't filled out a ticket yet but I am also experiencing this same issue on 2 printers since upgrading to 1.3.1.

closed 2017-02-15 by GitIssueBot