OctoPrint/OctoPrint #1743: Serial Port Dropdown Not Updating

scottmil, 2017-02-01:
#### What were you doing?
I was trying to connect Octoprint to a home-built  Arduino laser engraver (grbl-gcode) by specifying the Arduino serial port /dev/cu.wchusbserialfa140 in Settings->Serial Connection -> Additional Serial Ports using the glob /dev/cu.wchusbserial*  (see attached file Octoprint Settings.png).  

I also set the default port to Auto, then pressed the Save button.

BTW: I can successfully use Octoprint with my Printrbot Simple Metal 3D printer.  When I connect to my 3D printer, Octoprint successfully connects via /dev/ttyACM0 by Auto detect. (see attached Octoprint-Simple Metal.png)

#### What did you expect to happen?
I expected that the default port "Auto" plus the two Additional Serial Ports would show up in the main window Serial Port dropdown list.   The additional ports are listed in config.yaml (attached)

#### What happened instead?
An old serial port /dev/ttyUSB0 appeared as the dropdown default serial port, even though I had saved "Auto" as the default port.   /dev/ttyUSB0 also appears in config.yaml but I cannot delete it via the OctoPrint settings serial port settings.

#### Branch & Commit or Version of OctoPrint
1.3.1 master

#### Printer model & used firmware incl. version
Custom gcode laser engraver using Arduino Uno USB interface and Arduino grbl library 1.1e.20161208

#### Browser and Version of Browser, Operating System running Browser
Firefox 51.0.1 on Mac
Safari for iPad iOS 10.2.1

#### Link to octoprint.log
octoprint.log attached

#### Link to contents of terminal tab or serial.log
serial.log attached

#### Link to contents of Javascript console in the browser
Page source attached (page-source.rtf)

#### Screenshot(s) showing the problem:
OctoPrint Serial.png
OctoPrint Settings.png
OctoPrint-Simple Metal.png

[image]
[image]
[image]

[config.yaml.pdf](https://github.com/foosel/OctoPrint/files/743920/config.yaml.pdf)
[octoprint.log.pdf](https://github.com/foosel/OctoPrint/files/743921/octoprint.log.pdf)
[page-source.pdf](https://github.com/foosel/OctoPrint/files/743922/page-source.pdf)
[serial.log.pdf](https://github.com/foosel/OctoPrint/files/743923/serial.log.pdf)

I have read the FAQ.

foosel, 2017-02-15:
Sorry for the delay in replying to this, I was down sick the past two weeks and this ticket somehow slipped through when I was going through what came in during that time.

A quick question first: Did you reload the web interface after changing the list of additional ports? OctoPrint currently does not refresh the list and default of ports automatically, so that might be the reason why you were still seeing the other port here. You can install the [PortLister Plugin](http://plugins.octoprint.org/plugins/portlister/) to have OctoPrint actively scan in the background for changes in the list of available devices and then trigger the frontend to refresh that list.

Next is that OctoPrint by default will list anything that is available on your system _and_ matches any of the [default glob patterns](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/comm.py#L141-L147). The additional ports are really to be understood as additional. They do not overwrite the default list, they extend it.

So if you have both your printer and your laser arduino plugged in, you will see two ports (on refresh or automatically with the aforementioned plugin).

> When I connect to my 3D printer, Octoprint successfully connects via /dev/ttyACM0 by Auto detect. (see attached Octoprint-Simple Metal.png)

I'm utterly and completely surprised by that because OctoPrint's serial port auto detection (not the baudrate) works by attempting to set the controller into programming mode, and that at least didn't work the last time I attempted that with a Printrboard due to how the micro controller on those is wired up :flushed:

foosel, 2017-02-24:
Due to a lack of feedback to the contrary I'm assuming a missing reload (or PortLister plugin) was indeed the issue here. If not, feel free to request a reopen! :)

closed 2017-02-24 by foosel

scottmil, 2017-03-01:
Please accept my sincere apologies for the delay in providing feedback to issue  #1743 that I posted.

Following your recommendation, I installed the plugin PortLister, and indeed it updated the serial port list as you described when connected to my Printrbot Simple Metal.

When I connected OctoPrint to my laser engraver, however, OctoPrint was not able to detect the serial port (/dev/cu.wchusbserialfa140) even though I had added the glob pattern /dev/cu.wch* in Settings.   I suspect that this is not an OctoPrint issue but rather because the laser engraver use a Chinese-clone Arduino with a non-standard Winchiphead CH340/341 serial bridge chip.

Hope you are fully recovered from your recent illness.

Best regards,

Scott Miller
(a Patreon patron)

foosel, 2017-03-06:
> OctoPrint was not able to detect the serial port (/dev/cu.wchusbserialfa140) even though I had added the glob pattern /dev/cu.wch* in Settings. I suspect that this is not an OctoPrint issue but [...]

As mentioned above, OctoPrint will list anything in this dropdown that a) exists on the file system and b) is registered as a glob pattern. So if it didn't show up after plugging in the laser and refreshing the list (manually or through the PortLister) plugin, it sounds like the device was not hooked into the file system.

You could check if `dmesg` tells you anything about what the underlying Linux system decided to do with the laser engraver (e.g. if it even registered it as a serial device) and check if `ls /dev/cu.wch*` on a command line does return any results while the engraver is plugged in - the latter is basically what OctoPrint does too.

scottmil, 2017-03-22:
Hi Gina-

I just returned to the USA yesterday from several weeks in New Zealand, and today I had the opportunity to respond to your suggestion to verify the serial ports using the operating system file system.

Apparently the Chinese clone Arduino with the Winchiphead CH340/341 does not add a serial port entry in the /dev folder, so consequently it does show up in the Octoprint dropdown.  As a workaround, I added a port entry using this command:

sudo touch /dev/cu.wchusbserialportfa140

Now this port shows up in the dropdown without needing an Octoprint additional port glob pattern entry.

Thanks for your patience and assistance.

Best regards,

-Scott Miller

scottmil, 2017-03-22:
Actually, this is the best solution to my serial port connectivity (install a CH34x driver):

https://github.com/aperepel/raspberrypi-ch340-driver <https://github.com/aperepel/raspberrypi-ch340-driver>

Not an Octoprint issue at all!!